earlephilhower/arduino-pico #969: Access to PIO pins - WS2812 - New feature

Grumpy-Mike, 2022-11-08:
I am trying to access the PIO pins on a Pico2040. I know you have done this to emulate the software serial library, but I can't seem to find the vital information of how to do this from this repository, although I assume it the somewhere here. 
My aim is to use the PIO pins to drive a WS2812 LED string.

So is there any chance that could be added as a feature? Either access to the PIO or a driver for the WS2812.

earlephilhower, 2022-11-08:
There are no such things as PIO pins because the PIOs can control all pins.  Look at the `cores/rp2040/*.pio` files for some example PIO code and the SDK docs on PIO programming.  We include `pioasm` as part of the install, or you can use the online PIO assembler listed in the docs here.

Good luck!

closed 2022-11-08 by earlephilhower

sonnny, 2023-01-13:
@Grumpy-Mike Sorry if this is too late but I have a youtube demo about the subject, see link below

[youtube demo](https://www.youtube.com/watch?v=fFMufDrzliU)

Grumpy-Mike, 2023-01-16:
Hi sonnyl,
Thanks a lot for this.
No it is not too late, It still has time to affect the language choice for my current project.
My only stumbling block is that I can't get a USB implementation to work with this system, I am particularly concerned about making it a MIDI client.
None of the code I have tried either works or compiles.
At the moment I am using the Circuit Python MIDI client, but this has an odd fault in that it won't send any MIDI messages that start Cx, Dx, or Ex. I did rase an issue with Adafruit but it seems it works on later versions of the OS but not (for some unknown reason with OS 10.14.6).

A lot of people are stuck on this version because it is the last one that supports both 32 and 64 byte applications.  I have confirmed that it works on the later ones used on my iMac.

Anyway thanks a lot for this demo, I will try it shortly.

Cheers
Mike