earlephilhower/arduino-pico #971: MacOS Ventura 13.0.1 14:UNAVAILABLE : No Connection Established on Feather RP2040

wildestpixel, 2022-11-10:
Have setup IDE 2.0.1 on Ventura 13.0.1 installed as update overnight hoping to alleviate RP2040 uf2 issues.

When connected to the IDE on serial port /dev/cu.usbmodem.2201, with the Adafruit Feather selected as the board having installed support for it with this repo, the above error occurs - I cannot write anything to the board, verify or compile code.

Is there something obvious I am doing wrong ?

Many thanks in advance
[image]

fjansson, 2022-11-11:
There is a known problem with Ventura and the Pico:
https://www.raspberrypi.com/news/the-ventura-problem/
https://github.com/raspberrypi/pico-sdk/issues/1081

Not exactly what you report but I'd guess it's related. The problem is in copying the uf2 file to the Pico USB drive, which the Arduino IDE does behind the scenes.

wildestpixel, 2022-11-11:
If file copy is done behind the scenes, it doesn’t suffer from the above that was introduced in Ventura as its only a finder problem virtual usb drives. I hope that you are right in that it’s related, but a command line copy is fine for a uf2 as is a copy in a 3rd party app such as Forklift

maxgerhardt, 2022-11-11:
You can use this Arduino core with PlatformIO + VSCode too, and it will use a different upload method (via libusb instead of a plain file copy), that might allow you to work around this problem but still develop in VSCode: https://arduino-pico.readthedocs.io/en/latest/platformio.html

closed 2022-11-30 by earlephilhower