earlephilhower/arduino-pico #973: WiFi.localIP() hangs in Nano RP2040 Connect but fixed removing an analogRead() call (??)

stefat, 2022-11-13:
**Settings:**

Hw Mac Apple Silicon M1Pro
macOS Ventura 13.0.1
Sketch: Examples->WiFiNINA_Generic->WiFiWebServer
Arduino IDE 1.8.19
Nano RP2040 Connect sdk core: Arduino-Pico core in ~ /Library/Arduino15/packages/rp2040/2.6.2/
Libraries:
WiFiNINA_Generic 1.8.14-7
SPI 1.0 - from ~/Library/Arduino15/packages/rp2040/hardware/rp2040/2.6.2/libraries/SPI
**The Issue:**
Just successfully connected the sw appears hanging doing a IPAddress ip = WiFi.localIP() instruction.
The RP2040 connected as a Station or AP shows the same behaviour.

**The code:**

The following printWiFiStatus() is called: edited from the original example sketch.

void printWiFiStatus() 
{
  // print the SSID of the network you're attached to:
  Serial.print(F("SSID: "));
  Serial.println(WiFi.SSID());
  delay(1000);
  // print your board's IP address:
  Serial.print(F("IP load "));
  IPAddress ip = WiFi.localIP();
  Serial.print(F("IP Address returned:"));
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print(F("Signal strength (RSSI):"));
  Serial.print(rssi);
  Serial.println(F(" dBm"));
}`
The serial monitor shows:

18:01:17.764 -> Start WiFiWebServer on NANO_RP2040_CONNECT
18:01:17.764 -> WiFiNINA_Generic v1.8.14-7
18:01:18.488 -> Attempting to connect to SSID: TIM-86351700
18:01:38.388 -> SSID: TIM-86351700
18:01:39.373 -> IP load 
and nothing more.

**Additional info for a (very) strange fix**
The (very) strange thing is that in the sketch, removing an analogRead() call, fixes this issue.

the original code is: 
`// output the value of each analog input pin 
 for (int analogChannel = 0; analogChannel < 6; analogChannel++)  
 { 
   int sensorReading = analogRead(analogChannel); 
   client.print("analog input "); 
   client.print(analogChannel); 
   client.print(" is "); 
   client.print(sensorReading); 
   client.println("<br />"); 
 } `

and replacing this with e.g.:
`int sensorReading; 
  
 // output the value of each analog input pin 
 for (int analogChannel = 0; analogChannel < 6; analogChannel++) 
 { 
   // analogRead() will hang system in arduino-pico core 
   sensorReading = millis() % 1024;      //analogRead(analogChannel); 
   client.print("analog input "); 
   client.print(analogChannel); 
   client.print(" is "); 
   client.print(sensorReading); 
   client.println("<br />"); 
 } `

This does also the serial monitor output showing a correct IP address :
07:39:11.855 -> Start WiFiWebServer on NANO_RP2040_CONNECT
07:39:11.855 -> WiFiNINA_Generic v1.8.14-7
07:39:12.617 -> Attempting to connect to SSID: TIM-86351700
07:39:14.913 -> Attempting to connect to SSID: TIM-86351700
07:39:16.995 -> Attempting to connect to SSID: TIM-86351700
07:39:25.550 -> SSID: TIM-86351700
07:39:26.535 -> IP load IP Address returned:192.168.1.9
07:39:26.535 -> Signal strength (RSSI):-85 dBm

Any clue ?
Thanks for any help.

earlephilhower, 2022-11-28:
Without a self-contained MCVE sketch, there's not much we can do here.  Please pare it down to a simple example others can run to reproduce the issue.

JAndrassy, 2022-11-29:
analogRead(5) and analogRead(6) doesn't work on Nano Connect. only analogRead(A5) and analgRead(A6).
there are no 6 ADC pins on Nano Connect. A5 and A6 are controlled by the NINA module and from the RP2020 are accessible only as A5 and A6.

JAndrassy, 2022-11-29:
A4 ad A5 are I2C connected to the crypto chip and NINA module

earlephilhower, 2022-11-29:
Thanks for the explanation, @JAndrassy !  Makes sense, the pin mode gets borked since it's being used as `analogRead` here but really needs to be part of an I2C/etc. connection to the other onboard chips.

AcThPaU, 2023-03-13:
it's not the case for me, My A4 and A5 is used for I2C and it still hangs