earlephilhower/arduino-pico #975: Please add the Waveshare RP2040 Zero

IAmOrion, 2022-11-13:
Could you please add the Waveshare RP2040 Zero to the board list

[pins_arduino.zip](https://github.com/earlephilhower/arduino-pico/files/9997083/pins_arduino.zip)

maxgerhardt, 2022-11-14:
Schematics of Waveshare RP2040 Zero (https://www.waveshare.com/w/upload/4/4c/RP2040_Zero.pdf) say `W25Q16JVUXIQ` is used. You gave it `boot2_w25q080_2_padded_checksum` and it really worked?

IAmOrion, 2022-11-14:
> Schematics of Waveshare RP2040 Zero (https://www.waveshare.com/w/upload/4/4c/RP2040_Zero.pdf) say `W25Q16JVUXIQ` is used. You gave it `boot2_w25q080_2_padded_checksum` and it really worked?
> 
> If not, can you try [boot2_W25Q16JVxQ_4_padded_checksum](https://github.com/maxgerhardt/pico-boot2-generator/blob/master/generated/Winbond/W25Q16JVxQ/boot2_W25Q16JVxQ_4_padded_checksum.S)?

Bugger me, "woods for the trees" I reckon! Well found though, my apologies for missing that.  
Yes it's been working fine for me with `boot2_w25q080_2_padded_checksum `.  I will recreate the boards.txt now and test again and report back

maxgerhardt, 2022-11-14:
Okay, great. If The W25Q16.. file is used, that is also what CircuitPyhton is using (I do the same boot2.S generation script as them):

https://github.com/adafruit/circuitpython/blob/8f414eb4ee4db6489c0f9eca1e441200ff3e852a/ports/raspberrypi/boards/waveshare_rp2040_zero/mpconfigboard.mk#L9

So I think it's better that `boot2_W25Q16JVxQ_4_padded_checksum.S` is added used then.

IAmOrion, 2022-11-14:
I've updated my first post to mention your file and replaced the boot2 argument in my post to use you `boot2_W25Q16JVxQ_4_padded_checksum`

maxgerhardt, 2022-11-14:
The file you uploaded in the first comment is a zip that has a folder with a .h file and another zip, in which there is another zip.

I've opened the pull-request as linked above. Can you double check the variant header file is the latest one you uploaded?

IAmOrion, 2022-11-14:
@maxgerhardt Do'h - that'll teach me.  I zipped up the .h file, then realised I'd made a mistake, saved the changed .h file, and thought I should zip the folder too just because it matched what I put into makeboards.py - but I totally forgot to relete the zip that was in that folder! 

I shall double check my .h file later this evening and re-attach.  I shall remove the current zip and update later this evening

IAmOrion, 2022-11-14:
@maxgerhardt First post updated with new zip

[waveshare_rp2040_zero.zip](https://github.com/earlephilhower/arduino-pico/files/10007650/waveshare_rp2040_zero.zip)

maxgerhardt, 2022-11-15:
Thanks, PR updated.

closed 2022-11-21 by earlephilhower