earlephilhower/arduino-pico #976: ModuleNotFoundError: No module named 'encodings'

IAmOrion, 2022-11-13:
Just (re)installed this onto my Windows machine.  Latest v2.6.3.

No matter what board I choose, I always get the following Python error

```
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'
```

revell1, 2022-11-13:
@IamOrion, As you saw, I had the same issue.

My workaround is to:

1: UNINSTALL my current version of the library.
2: Exit the IDE.
3: Wait a minute or more, otherwise there appears to be some timing delay. Then open IDE.
4: Install the latest version of the library.
5: Exit the IDE.
6: Wait a minute or so again. Open IDE and build project.

This process has worked for me every time. Just updating a library used to work, but since about 2.6.1 this strange issue has appeared. I do not think I have updated the IDE version, so unclear why the issue has started.

IAmOrion, 2022-11-13:
@revell1 Yes I saw your post about it, for me, on Windows 11 22H2, that didn't work.  But as i say, using 2.6.1 works for me, and I don't think the changes affect my current projects so no big deal atm

fletchjp, 2022-11-18:
I am getting a similar error on a new computer with Windows 11 after moving my work from another computer.  I am wondering if there is a requirement to have Python installed. If so which version is needed?
Thanks
John

maxgerhardt, 2022-11-22:
What version of the Arduino IDE is in use here?

When you manually download and unpack https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.0.1-base/python3-3.7.2.post1-embed-win32v2a.zip, and start the python interpreter therein in a terminal and do `import encodings`, does it show the sammer?

fletchjp, 2022-11-23:
I am using Arduino IDE 1.8.19 and I install the software through the boards manager process. I am currently using Arduino-Pico 2.6.1 without any problems.

maxgerhardt, 2022-11-23:
I can't reproduce this in a Windows 11 VM directly from Microsoft. 
```
>ver
Microsoft Windows [Version 10.0.22000.1219]
```
and latest Arduino-Pico 2.6.3 installed via the board manager in Arduino IDE 1.8.19, downloaded in the .zip archive version, can compile sketches no problems.

[image]

* Do you have installed Python previously?
* Do you have python related environment variables, like PYTHON or PYTHONHOME set?
* Are you using the Arduino IDE as the .zip version or .exe or as a Windows App version?

earlephilhower, 2022-11-23:
Thanks @maxgerhardt for the assist!  I am also unable to reproduce on a Win10 physical system, nor on a temp VM I threw together.

@IAmOrion and @fletchjp if you could enable logging via `File->Prefs->Show Verbose Output->Compile` and rebuild with a failure and cut-n-paste the text output in the log window.  It looks like, for some reason, the IDE is not using the installed Python and somehow falls into using a system copy which is not the right 3.x version that is shipped w/the core.  This log would at least print out exactly what command line was being run when the failure happens.  If it's not the right Python being called, that would show.  Or, if there were missing params (we use "isolated" mode which should NEVER use anything but the distributed PyLibs) it would show it.

maxgerhardt, 2022-11-23:
I'm wondering if I have to go through a special sequence. The Windows 11 VM was fully windows-updated, but it was still downloaded some time ago and probably not what's not on https://developer.microsoft.com/de-de/windows/downloads/virtual-machines/ now. And I directly installed core 2.6.3, not 2.6.1 -> update to 2.6.3. I did however restart the Arduino IDE and compiled multiple times to make sure it would go through after program restarts.

IAmOrion, 2022-11-23:
@earlephilhower I shall enable the logging tomorrow and report back.

@maxgerhardt I had 2.6.1 installed (I do also have Python installed but thus far has never clashed or been an issue).  I installed 1.8.19 via the exe/msi installer direct from Arduino.cc.  I installed < 2.6.1 via the board manager.  At some point I updated to 2.6.1 and all was well.  I missed 2.6.2 purely because I didn't update the board when prompted.  I then started a new RP2040 project, and the board update was at 2.6.3, so I updated via board manager.  Kept getting the error mentioned, so I reverted back to 2.6.1 (again, skipping 2.6.2, purely because I knew 2.6.1 worked perfectly and I needed to compile and test stuff).

I have been using 2.6.1 since perfectly fine.  As mentioned I will enable logging and update to get more details on the error

maxgerhardt, 2022-11-24:
>I do also have Python installed but thus far has never clashed or been an issue

Is that the Windows Store version of Python or from https://www.python.org/downloads/windows/? Which version exactly?

fletchjp, 2022-11-24:
Earle, I have just seen this message, I have just shut down the computer where I have the problem and I won't be able to get to this for a bit - I am a Warley MRC member and involved with the Model Railway Show this weekend.

fletchjp, 2022-11-25:
Earle

I have come to the conclusion that the problem is not in either version 2.6.1. or 2.6.3. The problem is in the update process.

I have experienced the problem upgrading from 2.6.1 to 2.6.3 and downgrading back to 2.6.1. The error message about encodings was the same with both operations,  In each case the solution was to delete arduino-pico and do a reinstall of the required version. I can then successfully compile.

I hope this helps.

John

fletchjp, 2022-11-25:
I can send more information if you need it,

John

earlephilhower, 2022-11-26:
@fletchjp , looks like the Arduino IDE has a problem installing the core on your system for some reason and that might be enough to cause your failure (i.e. the Python extraction fails, causing the Python3.exe file to fail on startup with missing libraries).

Under Windows I've seen things like this with certain AntiVirus SW/anti-malware blockers where the scanner blocks creation of certain (legal, safe) filetypes because of the way it's running.  But in any case, it seems there's nothing wrong with the archive or the core here but with either the Windows IDE or unpacker which we don't have control over. :(

I would suggest you try the "uninstall, reinstall later version" option that was mentioned as working above.

fletchjp, 2022-11-26:
Earle  Thank you for this. My system is a fairly new Windows 11 setup where I am gradually installing new software. I have recently started to use the arduino-cli software which I had not used before. It was not running when I did these update experiments. I will have to be careful when I am next offered an update to Arduino-pico.

I am glad there is nothing wrong with your system.

Best wishes

John

IAmOrion, 2022-11-27:
Uninstall / Reinstall didn't work for me - 2.6.3 still errors.

Here's the full error message log - verbose output on compile is enabled

```
Arduino: 1.8.19 (Windows 10), Board: "Generic RP2040, 2MB (no FS), 133 MHz, Small (-Os) (standard), Disabled, Disabled, Disabled, Disabled, None, Pico SDK, IPv4 Only, Generic SPI /2"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\IAmOrion\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\IAmOrion\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\IAmOrion\Documents\Arduino\libraries -fqbn=rp2040:rp2040:generic:flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=picosdk,ipstack=ipv4only,boot2=boot2_generic_03h_2_padded_checksum -ide-version=10819 -build-path C:\Users\IAmOrion\AppData\Local\Temp\arduino_build_207126 -warnings=none -build-cache C:\Users\IAmOrion\AppData\Local\Temp\arduino_cache_265832 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-mklittlefs-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-gcc.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-gcc-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-python3.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-elf2uf2-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.4.0-c-0196c06 -verbose X:\Arduino Sketches\WireHarnessWrappingMachine_v1\WireHarnessWrappingMachine_v1.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\IAmOrion\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\IAmOrion\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\IAmOrion\Documents\Arduino\libraries -fqbn=rp2040:rp2040:generic:flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=picosdk,ipstack=ipv4only,boot2=boot2_generic_03h_2_padded_checksum -ide-version=10819 -build-path C:\Users\IAmOrion\AppData\Local\Temp\arduino_build_207126 -warnings=none -build-cache C:\Users\IAmOrion\AppData\Local\Temp\arduino_cache_265832 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pqt-mklittlefs.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-mklittlefs-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-mklittlefs\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-openocd-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-gcc.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-gcc-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-pioasm-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-pioasm\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-python3.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-python3-1.0.1-base-3a57aed.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed -prefs=runtime.tools.pqt-elf2uf2.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-elf2uf2\1.4.0-c-0196c06 -prefs=runtime.tools.pqt-elf2uf2-1.4.0-c-0196c06.path=C:\Users\IAmOrion\AppData\Local\Arduino15\packa [... cut ...]
Using library Adafruit_GFX_Library at version 1.11.3 in folder: C:\Users\IAmOrion\Documents\Arduino\libraries\Adafruit_GFX_Library 

Using library Adafruit_BusIO at version 1.14.1 in folder: C:\Users\IAmOrion\Documents\Arduino\libraries\Adafruit_BusIO 

Using library SPI at version 1.0 in folder: C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.6.3\libraries\SPI 

Using library Adafruit_SSD1306 at version 2.5.7 in folder: C:\Users\IAmOrion\Documents\Arduino\libraries\Adafruit_SSD1306 

exit status -1073740791

Error compiling for board Generic RP2040.
```

maxgerhardt, 2022-11-27:
>--publickey "X:\\Arduino Sketches\\

And it doesn't matter if your sketch is saved on the `C:\` drive, same error?

IAmOrion, 2022-11-27:
> > --publickey "X:\Arduino Sketches\
> 
> And it doesn't matter if your sketch is saved on the `C:\` drive, same error?

Correct.  I tried compiling from C: drive also, same error.  Just to re-iterate, v2.6.1 works perfectly using X: drive

maxgerhardt, 2022-11-27:
Can open a commandline (Win+R -> `cmd.exe`) and execute
```
C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed\python3
```
and if Python starts up correctly, execute `import encodings`.

Does it show any error?

What happens when you open another terminal, do `cd X:` first, then repeat the same commands?

IAmOrion, 2022-11-27:
> Does it show any error?

It doesn't start correctly - same error occurs

[image]

maxgerhardt, 2022-11-27:
Okay, that narrows it down further so the Python interpreter, and not to the `signing.py` file at all.

Close the Arduino IDE, execute 
```
explorer C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed\
```
then delete all files inside it, then download the Python version referenced in https://github.com/earlephilhower/arduino-pico/issues/976#issuecomment-1323554307, extract it in the folder so that it has the same files as before. Does startup still fail?

IAmOrion, 2022-11-27:
*BEFORE*
[image]

*AFTER*
[image]

[image]

So, that tells me that the problem is with the included Python as installed via the board manager - ?
To be clear, I had completely removed 2.6.1 including all folders and double checked all folder were gone in my arduino packages folder before I attempted installation of 2.6.3

Compilation *DOES* now SUCCEED!

maxgerhardt, 2022-11-27:
Wow so the Arduino IDE did not properly unpack the entire Python package.. because.. weirdness o_O? 

> To be clear, I had completely removed 2.6.1 including all folders and double checked all folder were gone in my arduino packages folder before I attempted installation of 2.6.3

So the entire `C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\` folder was deleted right?

IAmOrion, 2022-11-27:
> So the entire `C:\Users\IAmOrion\AppData\Local\Arduino15\packages\rp2040\` folder was deleted right?

Yes correct

maxgerhardt, 2022-11-27:
Wow, I could just reproduce the issue in the Windows 11 VM. Downgrading from 2.6.3 to 2.6.1 really showed the error immediately.

[image]

I'll file an issue in the Arduino IDE 1.x repo.

IAmOrion, 2022-11-27:
I guess this issue can closed then as it's not an issue with this, but an issue with the Arduino IDE -- at least we know why now :)

closed 2022-11-27 by IAmOrion

maxgerhardt, 2022-11-27:
This is now tracked in https://github.com/arduino/Arduino/issues/11842.

zorgoz, 2023-09-21:
Same issue with 3.5.0, same solution: 
Updating the content of 
`C:\Users\%USERNAME%\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed`
from this: https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.0.1-base/python3-3.7.2.post1-embed-win32v2a.zip

maxgerhardt, 2023-09-21:
I really think the issue is not with a particular version like 3.5.0, but the *update process* in the Arduino IDE itself. This will appear when updating to older and newer Arduino-Pico core versions alike. The best solution is to not handpick the missing files, but do a clean remove -> reinstall of the core.

zorgoz, 2023-09-21:
Remove-reinstall did not help. That was what I have tried first.

hrhg, 2024-03-02:
I also experienced the same problem.
To completely delete the rp2040 folder, I restarted Windows and then deleted it.