earlephilhower/arduino-pico #977: Blink program stops working when built with -DUSE_TINYUSB

delan, 2022-11-13:
Hi, thanks for making this really useful core!

I’m trying to get the TinyUSB stack working under PlatformIO, but whenever I add -DUSE_TINYUSB, my Pico no longer blinks or sends any text down Serial, though it still shows up at ttyACM0 and can still be flashed with pio upload. Minimal example below.

platformio.ini
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
build_flags = -DUSE_TINYUSB
; also tried -DCFG_TUSB_CONFIG_FILE=\"tusb_config.h\" with both
; 1. the upstream config linked in the docs https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/08c6ec947ae5671f66b9599214059ec73652ecdb/src/arduino/ports/rp2040/tusb_config_rp2040.h
; 2. the config in this repo https://github.com/earlephilhower/arduino-pico/blob/852219caf583f0d3e5ca7fe2c9c725f0f798019a/include/tusb_config.h
```

src/main.cc
```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
// also had <FreeRTOS.h> in my actual program

void setup() {
  // upload still works for me without this, despite the docs?!
  // https://arduino-pico.readthedocs.io/en/latest/usb.html#adafruit-tinyusb-arduino-support
  Serial.begin(115200);

  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  Serial.write("asdf ");
  Serial.flush();
}
```

pio pkg list
```
Resolving pico dependencies...
Platform raspberrypi @ 1.7.0+sha.fec7ed8 (required: git+https://github.com/maxgerhardt/platform-raspberrypi.git)
├── framework-arduino-mbed @ 3.1.1 (required: platformio/framework-arduino-mbed @ ~3.1.1)
├── framework-arduinopico @ 1.20602.0+sha.ec76310 (required: git+https://github.com/earlephilhower/arduino-pico.git#ec7631096b667b8d40663c6da4667e3e0aea5a97)
├── tool-mklittlefs-rp2040-earlephilhower @ 5.100300.220714 (required: earlephilhower/tool-mklittlefs-rp2040-earlephilhower @ ~5.100300.0)
├── tool-openocd-rp2040-earlephilhower @ 5.100300.220714 (required: earlephilhower/tool-openocd-rp2040-earlephilhower @ ~5.100300.0)
├── tool-rp2040tools @ 1.0.2 (required: platformio/tool-rp2040tools @ ~1.0.2)
└── toolchain-rp2040-earlephilhower @ 5.100300.220714 (required: earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714)
```

I have tried this on two Pico boards, one of which was brand new, and each on both my Linux machine and my partner’s Windows machine. Are you able to reproduce this at all? Thanks for your time.

delan, 2022-11-13:
Correction: it actually works on my partner’s Windows machine. But I also figured out the problem! I had another .cc file in src, that despite thinking I was no longer using it, was still being linked into the program. The offending line was

```c++
#include <FreeRTOS.h>
```

which also breaks on my partner’s machine. Have I missed some docs on how to use the core with both TinyUSB and FreeRTOS, or are the two features just not compatible, or…?

maxgerhardt, 2022-11-14:
Thanks for the report, I'll try and reproduce and debug it.

maxgerhardt, 2022-11-14:
I can reproduce your error, and I think it's a mistake in the core. The `__core0` FreeRTOS task, which is supposed to call into your `setup()` and `loop()`, is waiting for the variable `__usbInitted` to be `true`. 

https://github.com/earlephilhower/arduino-pico/blob/master/libraries/FreeRTOS/src/variantHooks.cpp#L60-L69

That variable is set to true in the "native Pico-SDK USB Stack" case when `__USBStart()` succeeds. 

https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/main.cpp#L106-L110

https://github.com/earlephilhower/arduino-pico/blob/master/libraries/FreeRTOS/src/variantHooks.cpp#L369-L376

But in the TinyUSB codepath, this variable is never set to `true`. And so, the task is infinitely blocked.

Let me see if I can do a simple modification to make this work..

maxgerhardt, 2022-11-14:
Using the quick fix linked above gets me up and running with TinyUSB + FreeRTOS.

[image]

Can I ask you to verify this by doing the same modification I did in the PR and reuploading? The file should be locally at `C:\Users\<user>\.platformio\packages\framework-arduinopico\libraries\FreeRTOS\src\variantHooks.cpp` lines 62 and 84.

delan, 2022-11-15:
That did the trick, thanks! I’m now having trouble getting reliable Serial output, but I can open a separate issue about that, because while it’s worse when -DUSE_TINYUSB and FreeRTOS.h are combined, it even happens with -DUSE_TINYUSB only.

zjwhitehead, 2022-11-16:
> but I can open a separate issue about that, because while it’s worse when -DUSE_TINYUSB and FreeRTOS.h are combined, it even happens with -DUSE_TINYUSB only.

I think I ran into something similar and ended up going a different direction. Perhaps I can help debug it. Add a link to the issue when you create it and Ill see if I can add any insights.