earlephilhower/arduino-pico #978: yaml-cpp link error undefined reference to `__atomic_fetch_add_4'

marnikvd, 2022-11-13:
Hello,

My aim is to put a yaml config file on the flash memory of the pico and parse it with the library [yaml-cpp](https://github.com/jbeder/yaml-cpp).  See also [here](https://github.com/jbeder/yaml-cpp/issues/592).
I already managed to solve several errors but the combination with [Adafruit_TinyUSB](https://arduino-pico.readthedocs.io/en/latest/usb.html) is difficult.

My test code, derived from [msc_external_flash.ino,](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/7264c1492a73d9a285512752b03f2550841c06bc/examples/MassStorage/msc_external_flash/msc_external_flash.ino)  contains this command :
`  config = YAML::LoadFile("/config.yaml");`
When I leave it out it compiles without errors.

With this line the compiler outputs :
`c:/users/myuser/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\myuser\AppData\Local\Temp\arduino-sketch-99EC6AB4379C04BB66AD81C894C7D53B\libraries\yaml\yaml-cpp\nodebuilder.cpp.o:
in function '_ZNSt13__atomic_baseIjE9fetch_addEjSt12memory_order': c:\users\myuser\appdata\local\arduino15\packages\rp2040\tools\pqt-gcc\1.4.0-c-0196c06\arm-none-eabi\include\c++\10.3.0\bits/atomic_base.h:548: undefined reference to ``__atomic_fetch_add_4'
collect2.exe: error: ld returned 1 exit status
`

I use arduino-pico version 2.6.3

My ideas to fix this have run out. Can someone help me?
Thanks

maxgerhardt, 2022-11-14:
That YAML-cpp library has pretty hard requirements on the runtime and compiler if it wants
```cpp
std::__atomic_base<unsigned int>::fetch_add(unsigned int, std::memory_order)
```
to be available (demangled symbol). Why does YAML-cpp need to be thread-safe with atomic usage when you only use it from one core and no RTOS is involved? 

I see the following options:
* Rewrite all usages of `std::atomic<T>` [here](https://github.com/jbeder/yaml-cpp/search?q=atomic) variables to just have the type `T`
* **If** the current compiler has a atomic library, then you *might* need to activate `-latomic` to link it in and resolve the undefined reference. You can control linking options by adding a `library.properties` file ot the `yaml` library with the `ldflags` field [as documented](https://arduino.github.io/arduino-cli/0.19/library-specification/)
* Add a C implementation of [`I4  __atomic_fetch_add_4  (I4 *mem, I4 val, int model)`](https://gcc.gnu.org/wiki/Atomic/GCCMM/LIbrary) yourself in a thread-unsafe manner (doesn't matter here anyways)
* If you find the compiler to be lacking `std::atomic` features, open an issue in https://github.com/earlephilhower/pico-quick-toolchain/, that's where the compiler comes from.

marnikvd, 2022-11-14:
Thank you very much for this comprehensive answer.
Nevertheless, since I am not a programmer, points 1 and 4 are beyond my capabilities (at the moment).
I tried point 2 but the result remains unchanged.  What I find strange is that I can't find a trace of `-latomic` in the compiler output.
Point 3 : I added your code snippet to my sketch, but I get compiler errors.  I didn't put much effort yet in finding the cause.

But there is still hope...
I started my project a few months ago and at that time I didn't find a YAML library for arduino.  Apperently, now there is [one](https://github.com/tobozo/YAMLDuino).  Perhaps I should check this out instead.

marnikvd, 2022-11-14:
this is the error for point 3
```
C:\Users\myuser\Documents\Arduino\test\msc_external_flash\msc_external_flash.ino:95:17: error: new declaration 'int __atomic_fetch_add_4(int*, int, int)' ambiguates built-in declaration 'unsigned int __atomic_fetch_add_4(volatile void*, unsigned int, int)' [-fpermissive]
   95 | extern "C" int __atomic_fetch_add_4(int *ptr, int val, int memmodel) {
      |                 ^~~~~~~~~~~~~~~~~~~~
```

marnikvd, 2022-11-14:
The compiler finds the function but the linker doesn't ??

maxgerhardt, 2022-11-15:
Can you add
```cpp
unsigned int __atomic_fetch_add_4(volatile void* ptr, unsigned int val, int memmodel) {
	(void)memmodel;
	unsigned int tmp = *(unsigned int*)ptr;
	*(unsigned int*)ptr = tmp + val;
	return tmp;
}
```
instead? 

>The compiler finds the function but the linker doesn't ??

Yes, the declaration of the function can be found but not its implementation.

marnikvd, 2022-11-15:
Now it compiles without errors.
Thanks again

closed 2022-11-15 by marnikvd