earlephilhower/arduino-pico #986: AdvancedWebServer example fails to compile for a Nano RP2040 Connect with Arduino-Pico version 2.6.3

stefat, 2022-11-17:
### Discussed in https://github.com/earlephilhower/arduino-pico/discussions/985

<div type='discussions-op-text'>

<sup>Originally posted by **stefat** November 16, 2022</sup>
**Settings:**

> Hw Mac Apple Silicon M1Pro
> macOS Ventura 13.0.1
> Arduino IDE 1.8.19
> Nano RP2040 Connect sdk core: Arduino-pico core rp2040 version 2.6.3

**The Issue:**
I'm using a Nano RP2040 Connect compiling the AdvancedWebServer example at this path:
~/Library/Arduino15/packages/rp2040/hardware/rp2040/2.6.3/libraries/WebServer/examples/AdvancedWebServer/AdvancedWebServer.ino
The IDE at compiling time does this error:
~/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.4.0-c-0196c06/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: ~/Documents/Arduino/BUILD DATA/sketch/AdvancedWebServer.ino.cpp.o: in function setup:
~/Documents/Arduino/SW CPU RP2040/SW core Arduino Pico/SA RP2040/TEST/AdvancedWebServer/AdvancedWebServer.ino:123: undefined reference to _ZN9WiFiClass5beginEPKcS1_
collect2: error: ld returned 1 exit status

**The code:**
The error is related at line 123 WiFi.begin(ssid, password) in the setup function:
`
void setup(void) {
pinMode(led, OUTPUT);
digitalWrite(led, 0);
Serial.begin(115200);
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
Serial.println("");
//........
}`
ssid and password have been set, as usual, with my current Wifi authentication.

Thank you for any idea on this.
</div>

JAndrassy, 2022-11-17:
for Nano RP2040 Connect you can't use the Pico W WiFi library and libraries around it.. you have to use the WiFiNINA library and the generic libraries around it

earlephilhower, 2022-11-21:
Thanks for the support, @JAndrassy

closed 2022-11-21 by earlephilhower