earlephilhower/arduino-pico #988: Problems with serial communication between Pico and Arduino Nano at 38400 bps.

vaelen, 2022-11-19:
I'm using a pico (RP2040) to read KISS packets from an Arduino Nano (ATmega28P). I'm using the earlephilhower/arduino-pico core to write my code.

Here is the important code:

```
SerialPIO console(2, 3);
SerialPIO tncSerial(0, 1, 1024u);

void init()  {
  console.begin(9600);
  tncSerial.begin(38400);
}

void loop() {
  while (tncSerial.available()) {
    console.write(tncSerial.read());
  }
}
```

In the physical wire up, I have the UART RX and TX pins from the Nano wired into a high speed level converter ([TXS0108E](https://www.ti.com/product/TXS0108E)) to reduce the voltage from 5V to 3.3V. On the 3.3V side of the converter, I have wired the RX/TX pins to GPIO 0 and 1 on the Pico. I also have the same pins, on the 3.3V side of the converter, wired to an FTDI UART cable so I can see what is being sent by the Nano. This should remove the level converter as a possible variable. I have also tried different GPIO ports on the pico to eliminate any specific ports as being an issue, and I have tried both hardware UARTs and PIO based UARTs. I also tried increasing the speed of the console to 38400, even though this shouldn't have any affect that I could think of, just to eliminate it as a possible variable. The FIFO on the input UART is large enough to hold the entire message that is coming from the Nano, so delays in reading should not cause an issue.

Here is the correct packet, as read directly from Nano:

```
c0 00 82 a0 92 9c 64 60  60 94 94 62 9e 96 82 60  |......d``..b...`|
ae 92 88 8a 62 40 62 ae  92 88 8a 64 40 63 03 f0  |....b@b....d@c..|
21 33 35 33 35 2e 32 32  4e 2f 31 33 39 32 39 2e  |!3535.22N/13929.|
34 30 45 2d 2f 41 3d 30  30 30 30 32 30 20 4e 6f  |40E-/A=000020 No|
67 61 79 61 2c 20 54 6f  6b 79 6f 20 2d 20 4d 65  |gaya, Tokyo - Me|
73 73 61 67 65 20 2f 20  57 69 6e 6c 69 6e 6b c0  |ssage / Winlink.|
```

Here is what I get on the pico side:

```
c0 00 82 a0 92 9c 64 60  60 94 94 62 9e 96 82 60  |......d``..b...`|
ae 92 88 8a 62 40 62 ae  92 88 8a 64 40 63 03 f0  |....b@b....d@c..|
21 33 35 33 35 2e 32 32  4e 2f 31 33 39 32 39 2e  |!3535.22N/13929.|
94 98 a2 96 97 a0 9e 98  98 98 98 99 98 90 a7 b7  |................|
b3 b0 bc b0 96 90 aa b7  b5 bc b7 90 96 90 a6 b2  |................|
b9 b9 b0 b3 b2 90 97 90  ab b4 b7 b6 b4 b7 b5 e0  |................|
```

The first 48 bytes are correct, but the rest are not. This looks like baud clock drift to me.
My guess is that the Nano is not able to generate a clock at exactly 38.4kHz and so the length of each bit is slightly off, causing the bits to drift in the signal. This seems to be corroborated by a [post I found](https://www.picotech.com/support/viewtopic.php?t=20971) discussing problems reading a Nano serial signal on a pico based signal analyzer. In this case they weren't using the pico's UART to read the data though, so it's slightly different but still related.

So here is my question: Do you think this is something that can be fixed in software? The FTDI chip reads the drifting signal fine, probably because it reads in the middle of each bit instead of at either the start or end of the bit, but this is just a hunch. I haven't dug into the code yet, but the fact that the hardware UART also fails to read it worries me.

Thanks for all you do! Let me know how I can help debug this farther.

earlephilhower, 2022-11-19:
Why not adjust the baud rate you're using on the Pico side?  The Nano may be at some off frequency, but it should be constant, no?

vaelen, 2022-11-19:
Oh, interesting. Can I give the pico any random baud rate?

vaelen, 2022-11-19:
Ok, after some further digging, I think something else is wrong. I flashed the nano with a very simple program that just prints a packet to the serial port every second and now the pico receives it without an issue. There must be something in the TNC firmware itself that is causing this. Closing the ticket as a red herring!

closed 2022-11-19 by vaelen