earlephilhower/arduino-pico #999: Example:  FSBrowser very very slow ; Response time very long > 60s

kendo55, 2022-11-23:
it seems that "webserver.streamFile" dos not work well. 

I use the original example "FSBrowser.ino" with LittleFS, 
the response time for e.g. the response of the call "serverIP/edit" is very very  long (> 60s).

The alternate test: 
*******************
With a replacemet in function **handleFileRead()**
 
of
````
 if (server.streamFile(file, contentType) != file.size()) {

````
with
````
    String sHTML = file.readString();
    file.close();
    DBG_OUTPUT_PORT.printf("HTML String bytes to send: %u\n", sHTML.length() );
    server.send(200, contentType, sHTML);
````

the Response time is very well.

it is clear that this does not work for larger content.
Just to test with the /edit-page

earlephilhower, 2022-11-29:
This comes down to the `ClientContext`'s `write(Stream)` method which is logically correct but very inefficient.
 https://github.com/earlephilhower/arduino-pico/blob/861da4c3618644b7e77b0ad8ad80d70ad760ff7c/libraries/WiFi/src/include/ClientContext.h#L372-L387

By sending bytes one at a time to LWIP, it may be generating a 50-byte packet for every single byte of data that needs to go out over WiFi and be ACK'd.  A simple buffered optimization would help speed things up here significantly, I think.

earlephilhower, 2022-11-29:
Running a sample speed test, this results in a >3x increase in transmit speed

````
// Placed in the public domain by Earle F. Philhower, III, 2022

#include <WiFi.h>
#include <LittleFS.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

int port = 4242;

WiFiServer server(port);

void setup() {
  Serial.begin(115200);
  LittleFS.begin();
  auto f = LittleFS.open("file.txt", "w");
  for (int i=0; i<10000; i++) {
    f.println("All work and no play make Jack a dull boy.");
  }
  f.close();
  WiFi.mode(WIFI_STA);
  WiFi.setHostname("PicoW2");
  Serial.printf("Connecting to '%s' with '%s'\n", ssid, password);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }
  Serial.printf("\nConnected to WiFi\n\nConnect to server at %s:%d\n", WiFi.localIP().toString().c_str(), port);

  server.begin();
}

void loop() {
  static int i;
  delay(1000);
  Serial.printf("--loop %d\n", ++i);
  delay(10);
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  uint32_t s = millis();
  auto f = LittleFS.open("file.txt", "r");
  client.write(f);
  f.close();
  client.flush();
  uint32_t e = millis();
  Serial.printf("Send time: %u ms\n", e - s);
}
````

Master:  `Send time: 8945 ms`
#1017 : `Send time: 2633 ms`

closed 2022-11-29 by earlephilhower

earlephilhower, 2022-11-29:
It's even way faster with `client.setNoDelay(true)` which is what the WebServer uses.

Adding in no delay to the above example and reducing it to 1,00 lines of text the results are **90x** faster:
master: `Send time: 21993 ms`
#1017: `Send time: 237 ms`

PR #1017 (2022-11-29): Speed up ClientContext::write(Stream) by chunking
Instead of sending a single packet per byte, send out larger chunks to TCP/IP while doing a ::write(Stream).

Fixes #999
commit d717b580 (2022-11-29): Speed up ClientContext::write(Stream) by chunking (#1017)

Instead of sending a single packet per byte, send out larger chunks to
TCP/IP while doing a ::write(Stream).

Fixes #999