earlephilhower/arduino-pico #1002: WiFi.firmwareVersion says "Please upgrade the firmware"

FraserElectronics, 2022-11-25:
I am using arduino-pico with my Pico-W and PlatformIO.
I want to test the WiFi functionality and use the ScanNetworks.ino code.
However, in the serial terminal I get "Please upgrade the firmware" and no WiFi networks are identified.
What am I supposed to do - is this asking for the WiFi firmware to be updated? 

Andy

earlephilhower, 2022-11-26:
I think you're using the Arduino Nano RP2040 version of the WiFi ScanNetworks code.  Arduino's not compatible with the CYW43 driver we have on the PicoW, so you need to use the example distributed here, https://github.com/earlephilhower/arduino-pico/blob/master/libraries/WiFi/examples/ScanNetworks/ScanNetworks.ino

closed 2022-11-26 by earlephilhower

FraserElectronics, 2022-11-26:
Thanks.
However, my Pico W still does not see any WiFi networks :-(

Andy

earlephilhower, 2022-11-26:
Have you selected the right board?  You need to specifically build for the `Raspberry Pi Pico W` in the boards menu.  Otherwise there's not WiFi code actually built. :)

FraserElectronics, 2022-11-26:
Thanks for pointing me in the right direction 👍
I am using platform.io so all I needed to change my board to rpipicow and it is working now :-)

Many thanks.

Andy