earlephilhower/arduino-pico #1010: wrong definition of LED for picow

5chufti, 2022-11-28:
the board definitions for rpipicow contain PIN_LED but on rpipicow the LED is not connected to a pin of RP2040 (but on wifi chip). 
So it is confusing that a successful compile and upload does not work as on plain rpippico (without w).

5chufti, 2022-11-28:
situation is not optimal (see edit above) but probably not easily solved, maybe some better division between examples for pico / picow

closed 2022-11-28 by 5chufti

kenneth558, 2022-11-30:
Do you know - LED_BUILTIN vs. PIN_LED: in scenarios where we want to sequester this pin from use by other devices in our designs, does this mean the Pico W frees up a pin for us that the Pico uses to control the LED?  And for either board, is there a difference between those two macros?  

Eventually, I'll want to know what to expect about this.  Thanks!

earlephilhower, 2022-11-30:
`LED_BUILTIN` is the default macro you should use.

The PicoW has several fewer available GPIOs because some are hardwired to the WiFi chip.  (It takes ~300KB worth of code to light that LED on the board because it's run by the 2ndary WiFi ARM processor.)  You can get more info from the Raspberry Pi Pico/Pico W datasheets.

kenneth558, 2022-12-01:
Thank you GREATLY for the education! I assume the datasheets will be most compatible with the standard Pico SDK rather than the Arduino IDE that I'm trying to remain with (perhaps ill-advisedly), hence my quicker resorting to post questions.  I'll follow your advice, though.

earlephilhower, 2022-12-01:
Sorry, not quite following.  For the pinouts (what's connected to the RP2040 microchip) you need the schematic which RPI provides as part of their docs.  For programming, you can use this core, the Pico SDK from RPI directly, MicroPython, etc.  They're orthogonal, really, but you do need to be sure you configure them for the right board (i.e. the LED_BUILTIN macro is different depending on which board you select from the menus.)