earlephilhower/arduino-pico #1014: LittleFS.remove(<filepath>) hangs with Pico W

stefat, 2022-11-28:
I'm running my project in a PicoW, just bought for testing, to be used as a local web server . 
rp2040 core is 2.6.3.

The LittleFS works well:
1. uploading from the IDE;
2. the root folder is correct. All files loaded;
3. browsing the root from the sw shows all files with their contents and the related website works.

The issue is that when the client asks for a delete operation on a file the code, running in Core0, hangs:
`
.........

  usb.print(F("  Deleting:<"));
  usb.print(filepath);
  usb.println(F(">"));
  //rp2040.idleOtherCore();
  bool success = LittleFS.remove(nome);
  if(success)
   {
       usb.println(F("  Deleted"));
    }
    else
    {
        usb.println(F("  Oops: failed"));
     }
........
` 
filepath contains the full path of the file. BTW in this case it's at the root level.

The Serial monitor shows:
19:55:19.989 -> 
19:55:19.989 ->   Deleting:</pippo.yrn>
and nothing more.
/pippo.yrn is the correct path of the file.
Core1 is idle from the boot. Just in case I tested also with another idleOtherCore method call just before the remove with no avail.

As a side note the FSUpload.ino and SpeedTest.ino examples work fine. But I did not find any example doing a file remove. 

Any clue ?
Thanks for any help on this.

earlephilhower, 2022-11-28:
Can you please make a small, self-contained MCVE that reproduces this?  Right now there's no way to debug anything since it's your own app and we only have a small bit available.

One general comment, make sure you are not deleting a file that you already have open as a `File` or else you will end up with hanging references and I'd expect it to crash eventually on `File` object destruction.

stefat, 2022-11-29:
Yes, you're right.
I've done the MCVE as per the attached  RemoveTest.zip.
[RemoveTest.zip](https://github.com/earlephilhower/arduino-pico/files/10112692/RemoveTest.zip)
The Remove method of the library works. Any file, with any suffix in the name, regardless it has been previously opened, appears correctly removed.

The cause is my environment sending a GET for page5.htm?del=filename. 
In my faulty sketch the summarized code is:
`
void setup()
{
//....
WiFiServer mWebServer(80);
WiFiClient client;
//...
}

void loop()
{
//....
client = mWiFiServer.available();
    if (client) {
      handleClient();
    }
//...
`
where: 
`
void handleClient() {
  char c;
  unsigned char count = 0;
  File webFile;                   // handle to files
  String req_str = "";            // buffer used to capture HTTP requests
  long data_length = -1;
  boolean skip = true;
  bool firstRequestlinePrinted = false;
  usb.println(F("### new client"));
  boolean currentLineIsBlank = true;
  while (client.connected())
  {
    if (client.available())
    {
        c = client.read(); // read 1 byte (character) from client
        req_str += c;
        if (c == '\n' && currentLineIsBlank)
        {
  //MARK: - ##### GET HOME AND OTHER html requests  ....
          if (req_str.startsWith("GET /index.htm") || req_str.startsWith("GET / "))
          {
            sendStandardPageHTTPResponseHeader(client);
            webFile = LittleFS.open("index.htm", "r");        // open home web page file
          }
          else if (req_str.startsWith("GET /page7.htm"))//System Reset
          {
              //......
          }
          else if (req_str.startsWith("GET /page5.htm"))
          {
              sendStandardPageHTTPResponseHeader(client);
              if (isUserAuthenticated >= 2)
              {
                int start = req_str.indexOf("&del=");
                if (start > -1)
                {
                    // qui una richiesta di delete di un file
                    int end = req_str.indexOf(" ", start+5);
                    String nome = req_str.substring(start+5, end); //nome del file da cancellare
                    bool success = LittleFS.exists(nome);
                    if(!success)
                    {
                      usb.println(F("  Oops: file does not exist !"));
                    }
                    else
                    {
                      usb.print(F("  Deleting:<"));
                      usb.print(nome);
                      usb.println(F(">"));
                      usb.flush();
                      bool success = LittleFS.remove(nome);
                      if(success)
                      {
                        usb.println(F("  Deleted"));
                      }
                      else
                      {
                        usb.println(F("  Oops: failed"));
                      }
                      usb.flush();
                    }
                }
                webFile = LittleFS.open("page5.htm", "r");        // re open the web page
              }
              else
              {
                webFile = LittleFS.open("forbid.htm", "r");        // open web page file
              }
          }
  //......
  //MARK: - ##### Send the web page/file to client
          if (webFile)
          {
               {
                //read webFile and does client.write(...) until the end of webFile
                //.....
                //.....
                }
                webFile.close();
                client.flush();
           }
          // reset all buffer elements
          req_str = "";
          break; // // end while (client.connected())
        }
        // every line of text received from the client ends with \r\n
        if (c == '\n')
        {
          // we're starting a new line
          currentLineIsBlank = true;
        }
        // we've gotten a character != '\r' on the current line
        else if (c != '\r')
        {
          currentLineIsBlank = false;
        }
      } // end if (client.available())
  } // end while (client.connected())
  delayMicroseconds(1000);      // give the web browser time to receive the data.
  client.stop(); // close the connection
  usb.println(F("### client disconnected"));
}

`
The home page and all the rest of the website, with Ajax requests, Post's etc. works.
The sw appears crashing just after the  bool success = LittleFS.remove(nome); line.
The filename (nome) is correct and appears in the serial monitor just before the - supposed - crash.
I realize it's not your job but ... any idea ? Just in case.
I'll anyway keep up investigating.
Thanks for your time.

earlephilhower, 2022-11-29:
Thanks for the MCVE, but I can't see a failure with it.

I just ran your attached INO and it works fine (other than the LED which needs a `pinMode(LED_BUILTIN, OUTPUT);` to actually blink (I'm running on a Pico, there is no difference between Pico/PicoW if there's no WiFi used).

I uploaded the `data` filesystem, ran, hit `Y`, and it went through the 3 files and printed the status messages w/o any incident and ended up in the `loop()` trying to blink the LED.  I double-checked using PicoProbe and GDB and everything is going fine on the system at the end...

Could you double-check that  your test sketch fails for you?

stefat, 2022-11-29:
I've not been clear, sorry. I confirm my MCVE works fine with me as well. The issue is in my original project.
I just give you an update: the LittleFS.remove doesn't work in my project because of an rp2040.idleOtherCore() just at the beginning of the startup() (!?). If I remove that, the LittleFS.remove works.
The funny(?) thing is that in the MCVE a rp2040.idleOtherCore() at the top of startup() does not affect the behaviour. The MCVE keeps working, always.
I have the suspect it's because I use many more libraries and resources (?) in my original project. 
See please attached my original main .ino of my project.
Sorry for the messy structure but it is a work in progress.
I've inserted in the startup() the same code of the MCVE (starting at line 344).
With line 151 rp2040.idleOtherCore(); commented out ... the file is correctly removed and the system works.
Removing the comment at line 151 the system hangs at line 357 bool success = LittleFS.remove("pippo.yrn");.

Funny issue, eh ?
[SA_4.0.ino.zip](https://github.com/earlephilhower/arduino-pico/files/10115448/SA_4.0.ino.zip)

earlephilhower, 2022-11-29:
>  The funny(?) thing is that in the MCVE a rp2040.idleOtherCore() at the top of startup() does not affect the behaviour. The MCVE keeps working, always.

Ah ha, that's the problem!

Why are you idling the other core?  The filesystem automatically protects itself by calling the same `idleOtherCore` (because when writing the flash, it's illegal to read from it by the other core...causes crashes) and you've just caused a deadlock.

I suggest just removing the `idleOtherCore` completely from your code...there's little need for it in 99.99% of apps.  If you do need it, the idea is to `idleOtherCore(); ...do some small dangerous bit of code...; resumeOtherCore()` (i.e. like an interrupt disable, only for a short time and always resume afterwards).

So, I think you're set.  Just drop the idling...

stefat, 2022-11-29:
Ahhhhhhhhhhh !!!!!!!!
LittleFS is too smart, by far, to me !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

BTW: this means that the multicore should work, shouldn’t it ?

Arrrrrrghhh ! Great !
I feel better now !
Thank you so much.

earlephilhower, 2022-11-29:
Multicore works as far as I know.

The gotcha is that Arduino libraries are not multicore/multithread safe, so you need to use them only on a single core.  WiFi should be run on core0.  The FS can be on either core, but you can't have both cores using the FS in parallel or it will go very badly.  Other Arduino libraries can probably be any core, but again only on that core you choose or else you'll wind up with really "interesting" crashes.

Good luck!

closed 2022-11-29 by earlephilhower