earlephilhower/arduino-pico #1027: FreeRTOS yield() and vTaskDelay(..) won't work on RP2040-Zero

jlarnal, 2022-12-03:
Hello everyone. 😃

Right now, I'm trying to understand why FreeRTOS `vTaskDelay(..)` or `yield()` both make the RP2040 hang on the _**RP2040-Zero**_ I'm working with ( arduino's `delay(..)` do the same btw ). 🤔

As I'm using an RP2040-Zero, the SWD isn't fanned out, so no proper debugging is available. 😤

Right now, the blinking is provided by the builtin WS2812, controlled through PIO1:SM3, and the delay is a recurrent call to `sleep_us(..)` on CORE0. ( from _./pico/time.h_ ) 🙄

But I didn't plan to have a blocking call such as `sleep_us(..)` here. I want `loop()`'s parent task to relinquish its time to the scheduler. And it just doesn't work.

Any info on a particular` yield()`/`vTaskDelay(..)` issue maybe ?
If I had the time, I'd throw some money at a pair of Pico boards to investigate this problem.

Here a snippet of code I just pruned from my project and tested in order to underline the issue : 

```

#include <stdint.h>
#include <FreeRTOS.h>
#include <task.h>
#include <Adafruit_TinyUSB.h>
#include <pico/time.h>
#include <NeoPixelConnect.h>

const unsigned char pixLvl[64] = {
  63, 57, 52, 47, 43, 39, 35, 32, 29, 26, 24, 22, 20, 18, 16, 15, 13, 12, 11, 10,
  9, 8, 8, 7, 6, 6, 5, 5, 4, 4, 4, 3, 3, 3, 3, 2, 2, 2, 2, 2, 2, 2, 1, 1, 1, 1, 1,
  1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1
};
NeoPixelConnect nPixL(16, 1, pio1, 3);  // uses PIO1's last state machine.

void setup() {
   nPixL.neoPixelClear();

   Serial.begin(115200);
   // delay(500); // Yeah no... hangs.. T_T
   Serial.print("Initializing...\n\tUSB serial coms ok\n"); // *tsk tsk... too early, won't show up on the VCP on time .
}

// Add the main program code into the continuous loop() function
void loop() {
   static unsigned long s_lastMs = 0;
   static unsigned char s_npix;

   if ((millis() - s_lastMs) > 1000) {
      s_lastMs = millis();
      Serial.printf("Elapsed :%u\n",millis() >> 10); // dividing by 1024 is wrong, but suits our needs. 
   }

   nPixL.neoPixelSetValue(0, pixLvl[s_npix] >> 2, 0, pixLvl[s_npix], true);
   s_npix = (s_npix + 1) & 63;

   /* You can uncomment exclusively ONE of theses two lines for a surprise. */
   sleep_us(32000);  // <--- "works" but obvisly block execution (puts the core in WFE)
   //vTaskDelay(32);   // <--- makes the RP2040-Zero device hang ! Why ?
   
}
```

maxgerhardt, 2022-12-06:
Are you using the Arduino IDE or PlatformIO? 

Does it happen with the even more simpler
```cpp
#include <FreeRTOS.h>
#include <Adafruit_TinyUSB.h>
#include <pico/time.h>

void setup() {
   Serial.begin(115200);
}

void loop() {
   Serial.println("Test"); Serial.flush();
   sleep_us(32000);  // <--- "works" but obvisly block execution (puts the core in WFE)
   //vTaskDelay(32);   // <--- makes the RP2040-Zero device hang ! Why ?
}
```
?

earlephilhower, 2022-12-09:
Cannot reproduce a failure with the simplified test, so closing unless you can repro w/o the library.
````
#include <FreeRTOS.h>
#include <task.h>
#include <pico/time.h>

void setup() {
   Serial.begin(115200);
}

void loop() {
   Serial.println("Test"); Serial.flush();
   //sleep_us(32000);  // <--- "works" but obvisly block execution (puts the core in WFE)
   vTaskDelay(32);   // <--- makes the RP2040-Zero device hang ! Why ?
}
````

Many Arduino libraries cannot be used w/FreeRTOS or any sort of multitasking or multithreading.  It's just not common on Arduino and there are no standard ways of handling things.  My guess is your library is not FreeRTOS compatible when task switches happen (which is what the `vTaskDelay` does) vs. a single task never exiting control of the core.

closed 2022-12-09 by earlephilhower