earlephilhower/arduino-pico #1028: Lingering pluggable_discovery.py processes on GNU/Linux

dos1, 2022-12-04:
For the past two days I have used arduino-cli to program an ESP8266 board while having RP2040 core installed. This resulted in many instances of RP2040's `pluggable_discovery.py` lingering on my system.

```
dos@pliszka:~$ ps aux | grep rp2040
dos      1914493  0.0  0.0  92520  1220 pts/85   Sl   gru02   1:02 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1914709  0.0  0.0  92520  1220 pts/85   Sl   gru02   1:02 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1914959  0.0  0.0  92520  1220 pts/85   Sl   gru02   1:01 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1915129  0.0  0.0  92520  1220 pts/85   Sl   gru02   1:01 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1915247  0.0  0.0  92520  1212 pts/85   Sl   gru02   1:02 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1915528  0.0  0.0  92520  1212 pts/85   Sl   gru02   1:02 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1915623  0.0  0.0  92520  1208 pts/85   Sl   gru02   1:01 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1915780  0.0  0.0  92520  1212 pts/85   Sl   gru02   1:02 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1916052  0.0  0.0  92520  1208 pts/85   Sl   gru02   1:02 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1916263  0.0  0.0  92520  1212 pts/85   Sl   gru02   1:02 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1979731  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:50 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1979964  0.0  0.0  92520  3248 pts/85   Sl   gru02   0:50 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1980516  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1980649  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:50 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1981181  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1981345  0.0  0.0  92520  1296 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1981438  0.0  0.0  92520  1496 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1981681  0.0  0.0  92520  1608 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1981753  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1981854  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1981978  0.0  0.0  92520  1436 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1982085  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1982189  0.0  0.0  92520  1212 pts/85   Sl   gru02   0:49 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      1993582  0.0  0.0  92520  1212 pts/85   Sl   gru03   0:47 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2142007  0.0  0.0  92520  1220 pts/87   Sl   gru03   0:23 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2142678  0.0  0.0  92520  1220 pts/87   Sl   gru03   0:23 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2142751  0.0  0.0  92520  1828 pts/87   Sl   gru03   0:23 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2143047  0.0  0.0  92520  1220 pts/87   Sl   gru03   0:23 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2149275  0.0  0.0  92520  1212 pts/87   Sl   gru03   0:22 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2149766  0.0  0.0  92520  1212 pts/87   Sl   gru03   0:21 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2219208  0.0  0.0  92520  1212 pts/87   Sl   00:17   0:10 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2219539  0.0  0.0  92520  1212 pts/87   Sl   00:19   0:10 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2224017  0.0  0.0  92520  1212 pts/87   Sl   00:37   0:10 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2224772  0.0  0.0  92520  1212 pts/87   Sl   00:40   0:10 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2225630  0.0  0.0  92524  1224 pts/87   Sl   00:42   0:10 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2225872  0.0  0.0  92520  1212 pts/87   Sl   00:43   0:09 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2227244  0.0  0.0  92520  1612 pts/87   Sl   00:48   0:09 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2227921  0.0  0.0  92520  1212 pts/87   Sl   00:50   0:09 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2228092  0.0  0.0  92520  1212 pts/87   Sl   00:51   0:09 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2228521  0.0  0.0  92520  1212 pts/87   Sl   00:52   0:09 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2290640  1.2  0.0  92520 10324 pts/91   Sl   12:47   0:00 python3 -I /home/dos/.arduino15/packages/rp2040/hardware/rp2040/2.6.3/tools/pluggable_discovery.py
dos      2290679  0.0  0.0   9120  2352 pts/91   S+   12:47   0:00 grep rp2040
```

Calling `arduino-cli board list` or any other command that triggers pluggable discovery process creates a new one and they never go away unless killed manually.

closed 2022-12-04 by earlephilhower

plurry, 2025-01-03:
I still have this problem on Arch running 4.4.1 and `arduino-cli`  1.1.1.

AqeeAqee, 2025-01-09:
Me too.
process Pytyhon3 for pluggable_discovery.py run out the CPU, I can hardly to anything.

- Windows 10
- Arduino 1.8.13
- arduino-pico 4.4.1

Walkaround: I renamed pluggable_discovery.py to prevent it excuting, or turn off Arduino-CLI in VSCode Arduino settings.

PR #1029 (2022-12-04): pluggable_discovery: Allow the scanner thread to quit
When receiving a `QUIT` message, the main thread was trying to tell    the scanner thread to quit - however, what it was actually doing was creating a local variable that shadowed the global flag used by the scanner thread. Fix that by ensuring that the main thread uses the global variable instead.

Fixes #1028
commit 4e16a700 (2022-12-04): pluggable_discovery: Allow the scanner thread to quit (#1029)

When receiving a `QUIT` message, the main thread was trying to tell   
the scanner thread to quit - however, what it was actually doing was
creating a local variable that shadowed the global flag used by the
scanner thread. Fix that by ensuring that the main thread uses the
global variable instead.

Fixes #1028