earlephilhower/arduino-pico #1031: Pico W connects to Wifi only every second boot

jokinghands, 2022-12-05:
Im trying to connect over C++ / Arduino IDE
If i take the Example code it would connect only every second boot. No matter if just flashed or Hardreset:

 WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
}

If I write it that way it would always conntect skipping the first try.

  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("STA: Failed!\n");
    WiFi.disconnect(false);
    delay(1000);
    WiFi.begin(ssid, password);
  }

it works with a while loop aswell. But he doesnt need more the a second attempt.

Anyone any Ideas? I guess its something about the router, i never experienced this with any ESP.

Thanks!

brentru, 2022-12-08:
Having a similar issue with the same code. @earlephilhower  Any ideas/advice for further debugging?

brentru, 2022-12-08:
Not running into the same issue with the `WiFiClient.ino` example, https://github.com/earlephilhower/arduino-pico/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino. Possibly an issue with setting `WiFi_STA`?

earlephilhower, 2022-12-09:
I've noticed that, depending on the state of the moon and whether or not it was already connected, the CYW43 chip onboard the Pico can take a long time to connect to an AP in my own setup.

My hunch is that you're just timing out for this reason.  Please add
````
WiFi.setTimeout(20000); // 20 sec to connect before giving up
````
before your `WiFi.begin()` and try again and report back, please.  I can always bump out the timeout if it's just a matter of that and others have even slower connections than myself...

StefanKellerAC, 2022-12-10:
Yes, it often fails "finally" with WL_CONNECT_FAILED.
I can reliably connect with this code after some seconds:

```
  int NTries = 99; 
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  int Status =  WiFi.status();
  while ( Status != WL_CONNECTED && NTries-- > 0) {
    Serial.printf ("NTries: %d, status: %d\n",NTries,Status);
    if (Status == WL_CONNECT_FAILED)
      {
      Serial.println("retry:");
      WiFi.begin(ssid, password);
      }
    else  
      delay(1000);
    Status = WiFi.status();
    }

  IPAddress myIP = WiFi.localIP();
  Serial.println(myIP);
```

washburnit, 2022-12-10:
I'm experiencing the same thing, most of the time it connects to my AP (Unifi UAP AC Lite) within 5/6 secs.
When it doesn't connect I have to reset it and then it connects.
On the "connected devices" page of the AP when it tries to connect it is displayed with its mac address as name, and stays there until it connects succesfully (then the name changes to "PicoW").

RickSeiden, 2022-12-11:
I don't know if this is relevant, but I noticed that it won't connect to serial until the serial monitor is open.  So if I have an

while (!Serial);

statement in my code, it never connects to the AP because it never gets past waiting for Serial to connect.

PR #1041 (2022-12-14): Increase WiFi.begin() default timeout to 15s
Fixes #1031, or at least covers the case of slower associations.
commit 03dbd6af (2022-12-14): Increase WiFi.begin() default timeout to 15s (#1041)

Fixes #1031, or at least covers the case of slower associations.
closed 2022-12-14 by earlephilhower