earlephilhower/arduino-pico #1032: Wrong wTotalLength for MIDI Interface Descriptor

0ba-pia, 2022-12-06:
Hi!

Probably a TinyUSB issue, not sure how to fix that but I had a report from a MIDI Host device manufacturer who told me that my Client MIDI device is not recognised because in the USB descriptor of the device, the wTotalLength is wrong:

Jacks:
07 24 01 00 01 25 00
06 24 02 01 01 00
06 24 02 02 02 00
09 24 03 01 03 01 02 01 00
09 24 03 02 04 01 01 01 00

Endpoints
09 05 01 02 40 00 00 00 00
05 25 01 01 01
09 05 81 02 40 00 00 00 00 
05 25 01 01 03

The given 0x0025 size is only the MS streaming descriptor and the jacks, but not the endpoints.
The correct value for wTotalLength must be 0x0025 +9+5+9+5 = 0x0041.
See paragraph B.4.2 in USB MIDI spec.

[image]

earlephilhower, 2022-12-06:
Good debug, but as you said this isn't the right spot to get it fixed.  I believe you're using Adafruit's TinyUSB library, so I'd re-post what you have put here in a new issue there for followup:  https://github.com/adafruit/Adafruit_TinyUSB_Arduino

If/when any updates are done there, we can pull in that new version here quickly.

closed 2022-12-06 by earlephilhower

0ba-pia, 2022-12-09:
Hi Earle,
I posted this issue to Adafruit's TinyUSB repository but didn't get any feedback yet.
Do you know where the issue might be from the code? If you point me to the right track I may be able to find the issue and try it out as this is pretty blocking to us. (BTW I am not hightly skilled at C++ as I am a Java guy; I looked at TinyUSB's code but I am afraid I am a bit lost).
Thanks.

earlephilhower, 2022-12-09:
I'm no expert in the MIDI or Adafruit wrapper side, but my best suggestion would be to look at `Adafruit_USBD_MIDI::makeItfDesc` in https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/5be0b21c0a5cf396b139bce0c6c0d403df051b0a/src/arduino/midi/Adafruit_USBD_MIDI.cpp

I think also you would get better help if you could make and post a simple MCVE sketch that makes the descriptor so that other people can at least repro your problem.   It wouldn't need to actually do MIDI, just get the descriptor wrong...

0ba-pia, 2022-12-13:
Hi Earle, I found the issue. It indeed comes from TinyUSB. A solution is here : [https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/223](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/223)  and I hope it'll be fixed soon.
Meanwhile I think this issue should remain open here until you pull the fix from Adafruit_TinyUSB lib.
Thanks for your help (and your great Arduino-Pico lib).

earlephilhower, 2022-12-14:
I've subscribed to that issue and when it gets merged I'll pull in the new lib from Adafruit.  (The Pico SDK will also need an update, but we don't export MIDI support in the current core from it so it doesn't matter.)