earlephilhower/arduino-pico #1033: Assigning new SPI pins disables the pico

NatanielSlo, 2022-12-07:
I needed to change the SPI pins i use for a SD card reader in my project. In the documentation i read that it could work like this:

```
void setup() {
    SPI.setRX(4);
    SPI.setTX(7);
    SPI.setSCK(6);
    SPI.setCS(5);
    SD.begin(5);
}
```

so i modified the SD library example and it turned out like this:

```

#include <SPI.h>
#include <SD.h>

const int chipSelect = 13;

void setup() {

  SPI.setRX(12);
  SPI.setTX(11);
  SPI.setSCK(10);
  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  Serial.print("Initializing SD card...");

  // see if the card is present and can be initialized:
  if (!SD.begin(chipSelect)) {
    Serial.println("Card failed, or not present");
    // don't do anything more:
    return;
  }
  Serial.println("card initialized.");
}

void loop() {
  // make a string for assembling the data to log:
  String dataString = "";

  // read three sensors and append to the string:
  for (int analogPin = 0; analogPin < 3; analogPin++) {
    int sensor = analogRead(analogPin);
    dataString += String(sensor);
    if (analogPin < 2) {
      dataString += ",";
    }
  }

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  File dataFile = SD.open("datalog.txt", FILE_WRITE);

  // if the file is available, write to it:
  if (dataFile) {
    dataFile.println(dataString);
    dataFile.close();
    // print to the serial port too:
    Serial.println(dataString);
  }
  // if the file isn't open, pop up an error:
  else {
    Serial.println("error opening datalog.txt");
  }
  delay(600);
}
```

Usually when i sent new code to my pico i just did and everything worked perfectly ( i could use the serial monitor etc. ). But after importing code with the defining pins  function it finishes but then disconnects so i am unable to open the serial monitor to see if it works. Also when i want to correct the code it doesn't show instantly in the ports selection menu but i need to press the boot select button as if it was the first time i plug it in. 

I am kind of new to programing so i am sorry if i post this in the wrong place but any help would be apprecieated.

earlephilhower, 2022-12-07:
The pinout you assigned is illegal.  Check the docs or https://cdn-learn.adafruit.com/assets/assets/000/099/339/original/raspberry_pi_Pico-R3-Pinout-narrow.png?1612915004 for legal pins for the SPI interface (the HW can only do certain combos and all the CK/DI/DO/etc. need to properly align).

closed 2022-12-07 by earlephilhower

NatanielSlo, 2022-12-07:
Thanks boss! It workes perfectly now.
Also if you don't mind anwsering is there an easy solution for using two SPI devices on different pins at the same time?  Can there be two different SPI's active with different pins defined or maybe can I change the pins every loop or maybe in the core there is something for this situation (i searched but couldn't find it).
Thanks again!

earlephilhower, 2022-12-07:
There are 2 SPI port, `SPI` and `SPI1`.  You can use the 2nd port, `SPI1` at the same time as `SPI`.  I don't think it's safe from the SPI protocol to change pins between transactions (glitches/etc. might upset the devices).