earlephilhower/arduino-pico #1039: Compiling error

fredykoci, 2022-12-12:
HI, When I try to build arduino sketch on rp2040 board compiler send error message about python modules. Python can not find encodings packages.

There is full message from compiler:

Arduino: 1.8.19 (Windows 10), Vývojová deska: "Raspberry Pi Pico, 2MB (no FS), 133 MHz, Small (-Os) (standard), Disabled, Disabled, Disabled, Disabled, None, Pico SDK, IPv4 Only"

Fatal Python error: initfsencoding: unable to load the file system codec

ModuleNotFoundError: No module named 'encodings'

Current thread 0x00002554 (most recent call first):

exit status -1073740791

Nastala chyba při kompilaci u desky Raspberry Pi Pico.

maxgerhardt, 2022-12-12:
>Fatal Python error: initfsencoding: unable to load the file system codec
>ModuleNotFoundError: No module named 'encodings'

Same error as in https://github.com/earlephilhower/arduino-pico/issues/976, please try cleanly uninstalling and reinstalling the Arduino-Pico core.

fredykoci, 2022-12-12:
Done Now I have a trouble with compiling code which is able to run when I compiling it on Mac but not on new installation Arduino and other libraries on PC. Compiling and Uploading is correct but after rp2040 restart it do not work and do not provide me com port and after that I have to force restart pico

fredykoci, 2022-12-12:
On mac 2.5.0 on PC also 2.5.0, and with this example it also works

maxgerhardt, 2022-12-12:
Can you upload the full sketch? Have you doubled checked that also all externall Arduino libraries used in the sketch are equal between Mac + PC?

earlephilhower, 2022-12-12:
Thanks for the assist, @maxgerhardt .  I'm closing this as a dup of #976 as you mentioned.

Very odd this started happening recently.  The ESP8266 core has the same Python ZIP file and I don't think I've heard any issues like this.

closed 2022-12-12 by earlephilhower