earlephilhower/arduino-pico #1040: PICO W ADC is only 10bit

AlanTwigg, 2022-12-14:
Using Arduino IDE 2.0.3 and a simple sketch to read the value of ADC2 - GP28.
Pin 34 ( ADC2 - GP28 ) is connected to 3.3V.
I am expecting a value of 4096, but am getting 1023

**Sketch**

int value = 0;
float voltage;
int ADC = 28;

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
 value = analogRead(ADC);

  Serial.print("Value = ");
  Serial.println(value);
  delay(500);
}

brabl2, 2022-12-14:
Default ADC resolution is 10bit. 
You need to set the 12-bit resolution: 
`analogReadResolution(12);`
More details in the documentation here:
https://github.com/earlephilhower/arduino-pico/blob/master/docs/analog.rst#void-analogreadresolutionint-bits

earlephilhower, 2022-12-14:
Thanks, @brabl2 .  @AlanTwigg we follow the Arduino default here:  https://www.arduino.cc/reference/en/language/functions/analog-io/analogread/

In any case, the Pico's ADC only has 8.7 ENOB so even at 10 bits you're not really losing anything...

AlanTwigg, 2022-12-14:
OK, many thanks, I must read the docs in the future!

closed 2022-12-14 by AlanTwigg