earlephilhower/arduino-pico #1048: WiFiServer example for Pico W not responding

SuperJonotron, 2022-12-18:
Wanted to test out the web server functionality so I loaded the example onto a new Pi Pico W. It connects to the network and I see the loop message in the console but if I try to go to the browser for the IP, the pico never responds. I see it making it to the pico as it reports:

`GET / HTTP/1.1`

But doesn't actually respond. Is the example broken? Some other mechanism required to get it to respond?

earlephilhower, 2022-12-18:
Sorry, can't reproduce a failure using the AdvancedWebServer.ino file.  What exactly are you running?  Remember you need to send two `\r\n`s to end the `GET` request if you're using `telnet`.  Also, unless you're using a BearSSL example, you may need to manually add `http://` to the URL in your browser since it won't be encrypted.

SuperJonotron, 2022-12-18:
I was using the WifiServer example which I expected to work with basic GET/POST requests but I guess this doesn't work that way.  I just tried telnet (never would have guessed that one) and it does appear to work with that type of connection.  Is this example only expected to be a telnet connection?  Didn't see that documented anywhere.

BearSSL example works as expected when tested.

earlephilhower, 2022-12-18:
`WiFiServer` isn't a HTTP server at all.  It just  shows how to use a TCP port connection (on 4242 so you'd have to manually set the port in a browser to even connect).  Lots of times you just need to get data in or out and it's way faster and lower memory just to use naked ports w/your own simplified protocol.

closed 2022-12-18 by earlephilhower