earlephilhower/arduino-pico #1053: Wire1 not working on SparkFun RP2040 Boards

SpectrumTech, 2022-12-19:
I'm trying to get a second I2C port working on both a SparkFun Thing Plus RP2040 and a SparkFun Pro Micro RP2040 in the Arduino environment (1.8.16 & 2.0).  

Using Wire.1 or something like:
  Wire1.setSDA(2);
  Wire1.setSCL(3);
  Wire1.begin();

causes the boards to freeze (i.e. will not contiunue past Wire.1 if something like Serial.println("foo"); follows, no output on the serial monitor, no output on any GPIOs set to blink LEDs, etc.).

Am I doing something wrong? Is this a known issue? Is there a fix or workaround?

I've tried communicating with various I2C devices and various values of pull up resistors (or none) and nothing seems to work.

Using "Wire" alone on the Qwiic port works just fine, but I need the second port for a particular reason.

Thanks for the help!

earlephilhower, 2022-12-19:
The following code runs fine:
````
#include <Wire.h>
void setup() {
  Serial.begin();
  delay(5000);
  Serial.printf("Start\n");
  Wire1.setSDA(2);
  Wire1.setSCL(3);
  Wire1.begin();
}

void loop() {
  Serial.printf("loop\n");
  delay(1000);
}
````

You'll need to provide an MCVE (small, self contained fully compile/runnable sketch) to get assistance.

earlephilhower, 2022-12-28:
Closing due to no update or MCVE

closed 2022-12-28 by earlephilhower