earlephilhower/arduino-pico #1054: freertos disturb pio work

scottxiaoyao, 2022-12-20:
```
void pio0_handler()
{
   gpio_set_mask( 1<<PIN_DEBUG_IO1 );  
    static uint8_t counter = 0;
    io_wo_32 *pData = NULL;
    if( vfd_state_machine.D )
    {
       pData = &SegmentList[counter].data[0];        
    }
    else
    {
       pData = &SegmentListBackUp[counter].data[0];  
    }
    pio0->txf[0] =  *(pData++);
    pio0->txf[0] =  *(pData++);
    pio0->txf[0] =  *(pData++);
    pio0->txf[0] =  *(pData);
    if(++counter == 40) counter = 0; 
    pio_interrupt_clear (pio0, 0);
    gpio_clr_mask(1<<PIN_DEBUG_IO1);    
}

void vApplicationTickHook(void){
  gpio_set_mask( 1<<PIN_DEBUG_IO0 ); 
  gpio_clr_mask( 1<<PIN_DEBUG_IO0);  
}

void initVFD_PIO(void){
    uint8_t sm = 0;
    uint offset = pio_add_program(pio0, &vfd_spi_program);
    vfd_spi_program_init(  pio0,              //PIO
                            sm,               //SM
                            16.625,           //16.625f,        //16.625f, 
                            offset,
                            PIN_SPI_DO,      
                            PIN_SPI_SCK, 
                            PIN_SPI_LAT,
                            PIN_BLK);
    pio_set_irq0_source_enabled(pio0,pis_interrupt0,true);
    irq_set_exclusive_handler(PIO0_IRQ_0, pio0_handler);
    irq_set_priority(PIO0_IRQ_0, 0);                     
    irq_set_enabled(PIO0_IRQ_0, true);
    pio0_handler();
}

```
[image]

when i include FreeRTOS.h and task.h

[image]

PIN_SPI_SCK SINGAL is wrong

earlephilhower, 2022-12-20:
Sorry, you will need to explain what exactly the issue you're seeing is, and provide a MCVE(mini working complete example).  FreeRTOS can swap things out even during an interrupt, so if you're seeing that it's expected and you need to protect the IRQ with the appropriate FreeRTOS calls to avoid task swapping...

scottxiaoyao, 2022-12-20:
thank you

pio.h
```
// -------------------------------------------------- //
// This file is autogenerated by pioasm; do not edit! //
// -------------------------------------------------- //

#pragma once

#if !PICO_NO_HARDWARE
#include "hardware/pio.h"
#endif

#define vfd_spi_wrap_target 0
#define vfd_spi_wrap 7

static const uint16_t vfd_spi_program_instructions[] = {
            //     .wrap_target
    0xbf42, //  0: nop                    side 7 [3] 
    0xb742, //  1: nop                    side 5 [3] 
    0xa442, //  2: nop                    side 1     
    0x6001, //  3: out    pins, 1         side 0     
    0x04e3, //  4: jmp    !osre, 3        side 1     
    0xc400, //  5: irq    nowait 0        side 1     
    0xb742, //  6: nop                    side 5 [3] 
    0xbf42, //  7: nop                    side 7 [3] 
            //     .wrap
};

/*
.program vfd_spi
.side_set 3

.wrap_target
    nop                 side  7 [3]     
    nop                 side  5 [3]  
    nop                 side  1       
bitloop:            
    out pins, 1         side  0 
    jmp !osre bitloop   side  1 
    irq 0                side  1 
    nop                 side  5 [3]
    nop                 side  7 [3]
.wrap
*/

#if !PICO_NO_HARDWARE
static const struct pio_program vfd_spi_program = {
    .instructions = vfd_spi_program_instructions,
    .length = 8,
    .origin = -1,
};

static inline pio_sm_config vfd_spi_program_get_default_config(uint offset) {
    pio_sm_config c = pio_get_default_sm_config();
    sm_config_set_wrap(&c, offset + vfd_spi_wrap_target, offset + vfd_spi_wrap);
    sm_config_set_sideset(&c, 3, false, false);
    return c;
}

static inline void vfd_spi_program_init(PIO pio, 
                                        uint sm,
                                        float clkdiv, 
                                        uint offset,
                                        uint pin_data,
                                        uint pin_sck, 
                                        uint pin_lat,
                                        uint pin_blk
                                        ) 
{
    pio_sm_set_consecutive_pindirs(pio, sm, pin_data, 1, true);
    pio_sm_set_consecutive_pindirs(pio, sm, pin_sck, 3, true);

    pio_gpio_init(pio, pin_data);
    pio_gpio_init(pio, pin_sck);
    pio_gpio_init(pio, pin_lat);
    pio_gpio_init(pio, pin_blk);

    pio_sm_config c = vfd_spi_program_get_default_config(offset);
    sm_config_set_out_pins(&c, pin_data, 1);
    sm_config_set_sideset_pins(&c, pin_sck);
    sm_config_set_out_shift(&c, true, true, 32);

    sm_config_set_clkdiv(&c, clkdiv);

    pio_sm_init(pio, sm, offset, &c);

    pio_sm_set_enabled(pio, sm, true);
}

#endif
```

main.ino
```
#include "pico/stdlib.h"
#include "pio.h"

#include <FreeRTOS.h>
#include <task.h>

#define  PIN_SPI_DO     2
#define  PIN_SPI_SCK    3
#define  PIN_SPI_LAT    4
#define  PIN_BLK        5

#define PIN_DEBUG_IO0   0
#define PIN_DEBUG_IO1   1

void pio0_handler()
{
    gpio_set_mask( 1<<PIN_DEBUG_IO1 );  //set goio1 high
    pio0->txf[0] =  0xAAAAAAAA;
    pio0->txf[0] =  0xAAAAAAAA;
    pio0->txf[0] =  0xAAAAAAAA;
    pio0->txf[0] =  0xAAAAAAAA;
    pio_interrupt_clear (pio0, 0);
    gpio_clr_mask(1<<PIN_DEBUG_IO1);    //set goio1 low
}

void initVFD_PIO(void)
{
    uint8_t sm = 0;
    uint offset = pio_add_program(pio0, &vfd_spi_program);
    vfd_spi_program_init(  pio0,              //PIO
                            sm,               //SM
                            16.625f,            
                            offset,
                            PIN_SPI_DO,      
                            PIN_SPI_SCK, 
                            PIN_SPI_LAT,
                            PIN_BLK);
                            
    pio_set_irq0_source_enabled(pio0,pis_interrupt0,true);
    irq_set_exclusive_handler(PIO0_IRQ_0, pio0_handler);
    irq_set_priority(PIO0_IRQ_0, 0);                     
    irq_set_enabled(PIO0_IRQ_0, true);
    pio0_handler();
}

void setup() {
    gpio_init(PIN_DEBUG_IO0);
    gpio_set_dir(PIN_DEBUG_IO0, GPIO_OUT);
    gpio_init(PIN_DEBUG_IO1);
    gpio_set_dir(PIN_DEBUG_IO1, GPIO_OUT);

    initVFD_PIO();
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1);
}

void setup1() {
  // put your setup code here, to run once:
}

void loop1() {
  // put your main code here, to run repeatedly:
}

void vApplicationTickHook(void)
{
  gpio_set_mask( 1<<PIN_DEBUG_IO0 );  //set goio0 high
  gpio_clr_mask( 1<<PIN_DEBUG_IO0);   //set goio0 low
}

```

pio interrupt   push 4*32bits data to pio fifo, pio  out all data to PIN_SPI_DO  then  call pio_irq and wait data to fifo,
First, the pio interrupt sets the GPIO1 to high level,and then push data to fifo ,and finally set GPIO1 to low level,
In fact, when the gpio1 is set to a low level, all the data enters the fifo, and the pio will take over the output of the data. At this time, the freertos task scheduling should not affect the work of the pio.

maxgerhardt, 2022-12-20:
And the signals always look okay when you remove FreeRTOS from the sketch? I.e. no
```cpp
#include <FreeRTOS.h>
#include <task.h>

void vApplicationTickHook(void)
{
  gpio_set_mask( 1<<PIN_DEBUG_IO0 );  //set goio0 high
  gpio_clr_mask( 1<<PIN_DEBUG_IO0);   //set goio0 low
}
```

scottxiaoyao, 2022-12-20:
> And the signals always look okay when you remove FreeRTOS from the sketch? I.e. no
> 
> ```c++
> #include <FreeRTOS.h>
> #include <task.h>
> 
> void vApplicationTickHook(void)
> {
>   gpio_set_mask( 1<<PIN_DEBUG_IO0 );  //set goio0 high
>   gpio_clr_mask( 1<<PIN_DEBUG_IO0);   //set goio0 low
> }
> ```

yes  always okay.

maxgerhardt, 2022-12-20:
With FreeRTOS enabled, does it help to add `noInterrupts()` at the start and `interrupts()  at the end of the `pio0_handler()` function?

earlephilhower, 2022-12-20:
I'd also recommend disabling preemption, too.  Your code can be preempted even inside an IRQ because the task switch happens via HW exception (which is above all IRQs).
https://freertos.org/symmetric-multiprocessing-introduction.html#smp-specific-apis

````
vTaskDisablePreemption(...)
noInterrupts();
<do work
interrupts();
vTaskEnablePreemption(...);
````

scottxiaoyao, 2022-12-21:
Issues remains, I don't think the pio0_handler() was interrupted

```
void pio0_handler()
{
    noInterrupts();
    gpio_set_mask( 1<<PIN_DEBUG_IO1 );  //set goio1 high
    pio0->txf[0] =  0xAAAAAAAA;
    pio0->txf[0] =  0xAAAAAAAA;
    pio0->txf[0] =  0xAAAAAAAA;
    pio0->txf[0] =  0xAAAAAAAA;
    pio_interrupt_clear (pio0, 0);
    gpio_clr_mask(1<<PIN_DEBUG_IO1);    //set goio1 low
    interrupts();
}
```

[image]

maxgerhardt, 2022-12-21:
Can the glitch also be observed at lower PIO frequencies? A clock divider of only `16.625f` at a 133MHz results in a 8.0 MHz signal (well, if floating points were accurcate, of course..), I don't think my logic analyzer is fast enough to detect an even smaller glitch there.

Another thought I had was that maybe the internal RP2040 clocks were messed with by FreeRTOS (like, readjusted) and that effects the PIO's clock source too, but in the configured mode (*not* tickless idle) it shouldn't be doing that..

earlephilhower, 2022-12-21:
Could this be the PIO trying to pull on an empty FIFO and stalling?  You've enabled it and if the FIFO is empty the `out` is going to wait until data gets into the FIFO.

All the RP2040 specific stuff is in `libraries/FreeRTOS/lib/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c` and `libraries/FreeRTOS/src/variantHooks.c`, none of which touch any clocks anywhere AFAICT.,

To simplify things, and discard the notion of the PIO clocks freezing, can you make a sketch w/FreeRTOS.h included and just a simple `pinMode(0, OUTPUT); tone(0, 10000)` and either capture it or hook it to a speaker.  Missed cycles will be easy to hear or spot/decode by the analyzer. 

Another alternative is this is a glitch on the analyzer side.  I'm not sure what your SCK is or your sampling frequency, but you obviously need at least ~8x sampling vs. clock rate or you may miss transitions.

scottxiaoyao, 2022-12-22:
sampling frequency is 200Mhz

earlephilhower, 2022-12-23:
There is no PIO clocking issue and I can't reproduce any hiccups using a plain, simple PIO program:
````
#include <FreeRTOS.h>

void setup() {
    pinMode(16, OUTPUT);
    tone(16, 60000);
}

void loop() {
}
````

Gives a clean 100khz waveform to the limits of my poor logic analyzer, with jitter less than 2 sampling periods (1/24MHZ) over a 5s test.

I also messed around with your code and can repro the dropped changes with the version you pasted.  What is interesting is that if you drop the GPIO toggle from either the PIO IRQ or the AppTask IRQ/exception handler things work.  Also, the PIO is continuing to shift out data just fine, just one pin is corrupted.  

Your problem seems to be that your code has **both** cores writing to the **single** GPIO-SET or GPIO-clear HW register at the same time as the PIO which (I'd assume) is illegal.  I suggest protecting the pin toggles with a FreeRTOS mutex or drop the application tick one.

closed 2022-12-23 by earlephilhower