earlephilhower/arduino-pico #1056: Hardware uart serial.read() on a no device returns a hidden character

drakejest, 2022-12-21:
I was running some test and noticed that when no uart devices are connected to the raspberry pi pico and saving the response into a string generates a hidden character

To replicate this setup you just need a raspberry pi pico and do not connect anything on it and run this code
```
#define device1_TX  0
#define device1_RX  1
#define device2_TX  4
#define device2_RX  5

void setup() {
  Serial1.setRX(device1_RX);
  Serial1.setTX(device1_TX);
  Serial1.setFIFOSize(512);
  Serial2.setRX(device2_RX);
  Serial2.setTX(device2_TX);
  Serial2.setFIFOSize(512);

  Serial.begin(115200);
  Serial1.begin(9600);
  Serial2.begin(9600);

  while (!Serial);

  String uartResponse;

  Serial.println(uartResponse.length());
  uartResponse = Serial1.readStringUntil('\r');
  Serial.println(uartResponse.length());

  String uartResponse2;

  Serial.println(uartResponse2.length());
  uartResponse2 = Serial2.readStringUntil('\r');
  Serial.println(uartResponse2.length());
}

void loop() {
}
```

This outputs in the serial monitor

```
22:21:31.260 -> 0
22:21:31.260 -> 1
22:21:31.260 -> 0
22:21:32.277 -> 1
```

I have tried this with `Serial1.read()` and `Serial1.readString()`. 

Things i have noticed is that this is a problem for the hardware UART only. The PIO based uart does not generate these characters.

earlephilhower, 2022-12-21:
Check the latest release or the `git` head.  There was a bugfix that drops received chars with framing/etc. errors.

Also, FWIW, you don't want to leave an input pin unconnected as you may end up catching noise.  The latest version will drop those chars, but it's generally a bad idea to have floating inputs...

drakejest, 2022-12-21:
> Check the latest release or the `git` head. There was a bugfix that drops received chars with framing/etc. errors.
>
> Also, FWIW, you don't want to leave an input pin unconnected as you may end up [...]

Oh there is a new version.

Tested and it not there anymore. Thank you !

I agree with you, i did not intend to leave it hanging, i uploaded a new code without the uart device connected, and  just happen to catch during the device check it return a single character

Anyway... 
Thanks for the hard work as always :)

closed 2022-12-21 by drakejest