earlephilhower/arduino-pico #1065: Feature request: Add MCLK support to I2S

DatanoiseTV, 2022-12-24:
Some I2S codecs require a MCLK signal which is 64/128/256*fs. This signal needs to be aligned with the bit clock (BCLK).
Adding master MCLK support would extend the support to many more widely available codecs (like the WM8731).

earlephilhower, 2022-12-24:
I think you could do this by using a 2nd PIO SM with a 2insn clock generator, setting the clock divider == the I2S PIO clock << 8, and starting both PIOs with a single write.

Back of the envelope 44.1khz * 256 * 2 = 22.6MHz so doable I think.

DatanoiseTV, 2022-12-25:
I think so as well, but the BCLK and MCLK need to be sync'd. Maybe this could be done by sampling the MCLK in one SM and couting the pulses.

jfrey-xx, 2022-12-25:
Hello, I step in following my own experimentations as described there https://github.com/earlephilhower/arduino-pico/issues/647#issuecomment-1304944161 and the code I use there https://gist.github.com/jfrey-xx/37c645b0fa87d86375f8a356104951d9

From what I have seen in the datasheet of various I2S chips, MCLK doesn't need to be in phase with other signals, hence  you might not need to start it _exactly_ at the same time as the other clocks. At least it's okay with the SGTL5000 I'm interfacing the pico with (teensy audio board). My only issue with generating MCLK is that I need to change the CPU clock to a multiple of the desired frequency to avoid pops.

@DatanoiseTV, do you have some I2S codecs you could test with my gist, to see if this way of generating MCLK works beside wy own case, and if that could serve as a first patch?

DatanoiseTV, 2022-12-25:
> @DatanoiseTV, do you have some I2S codecs you could test with my gist, to see if this way of generating MCLK works beside wy own case, and if that could serve as a first patch?

Not yet. I have some boards coming up in january and can test it then.
I am wondering if it would be feasible to implement ADAT 8 channel audio.

palmerr23, 2023-06-16:
It appears that Robin Playduck may have already resolved the MCLK issue using the Waveshare code as an example.

See lines 82-96 in https://github.com/playduck/pico-dsp/blob/master/src/pio_i2s.pio 

Waveshare uses another coding approach toa achieve the same outcome: https://www.waveshare.com/w/upload/9/92/Pico-Audio-V2.7z

I'll test the playduck additions to the Pico I2S.pio core and report back.

earlephilhower, 2023-06-16:
Neat idea.  To synchronize multiple PIOs they're doing a `WAIT IRQ`.  Otherwise, looks a pretty straightforward addition to the code here.

earlephilhower, 2023-06-16:
Might also be able to just inject the `WAIT` instruction without changing the existing PIO code.  That would save a couple insn words (and they're pretty precious since we only have 32 per PIO).

palmerr23, 2023-06-21:
Turned out to be a 2 instruction PIO program. There should be no need to synchronize the SMs as they are already deterministic, and the exact MCLK/BCLK phase isn't generally important as long as it doesn't shift over time.

```
.program pio_i2s_mclk		; RP
; PIO SM clock is MCLK x 2
; MCLK/BCLK sync is maintained by ensuring IN/OUT PIO SM clock dividers 
; are x128, 384, 256, 512 or 768 multiples of MCLK SM divider, as required.

loop_mclk:
	set pins, 1 ; toggle mclk
	set pins, 0 
	; Loop back to beginning...
```
All I needed to do was to make the BCLK clock divider a proper multiple of the MCLK divider.
```bool I2S::enableMCLK(int mult) {
    if (_running || !_isOutput) {
        return false;
    }
    _MCLKenabled = true;
    if ((mult % 64) != 0 || mult < 128)
        return false;
    _multMCLK = mult;
    return true;
}
```

I had to change the pio_sm_set_clkdiv() to the int_frac version with zero for the fraction.

Then just set MCLK to samplerate * multiplier.

MCLKbegin() is conditionally called from begin() (after the setFrequency() call) in this version, just to keep the new code separate.
```
void I2S::MCLKbegin() {  // RP
    int off = 0;
    _i2sMCLK = new PIOProgram(&pio_i2s_mclk_program);
    _i2sMCLK->prepare(&_pioMCLK, &_smMCLK, &off); // not sure how to use the same PIO
    pio_i2s_MCLK_program_init(_pioMCLK, _smMCLK, off, _pinMCLK);
    int mClk = _multMCLK * _freq * 2.0 /* edges per clock */;
    pio_sm_set_clkdiv_int_frac(_pioMCLK, _smMCLK, clock_get_hz(clk_sys) / mClk, 0);   
    pio_sm_set_enabled(_pioMCLK, _smMCLK, true);
}
```

[image]

24 bit, 48kHz * 256 multiplier. The clock MCLK alignment looks a little shaky, but that's because I only own a 200MHz logic analyser.

I have it working up to 192 * 48kHz on a PCM5102A in 24 bit mode before exceeding the DAC's maximum MCLK speed. MCLK 48kHz * 384 has a clock divider of 1 (147.6 MHz) and seems to work - but my logic analyser can't verify that with any accuracy!

Two calls are needed in the mainstream code - I might reduce this to just setMCLK(), as it can do both tasks.

I also had to set sys_clk to something sensible, so that integer clock dividers would work. sys_clk = 147.6 MHz turns out to be right for all 48kHz formats; and 135.6 for 44.1.
```
  set_sys_clock_khz(147600, false);  
  i2s.setMCLK(pMCLK);  // set the MCLK pin 
  i2s.enableMCLK(384); // samplerate -> MCLK multiplier
```

I'll do some more checking at different bit widths, sample rates and MCLK multipliers, tidy up the code and do a pull. 

I'll also add some entries and extra notes for the doco.

palmerr23, 2023-06-21:
Here's the output spectrum - no evidence of clock jitter.
[image]

palmerr23, 2023-06-21:
I also noticed that setFrequency() should probably use
`int bitClk = _freq * (_bps +1) * 2.0 /* channels */ * 2.0 /* edges per clock */;`
in Phillips (not LJ) mode, as there's an extra BCLK per LRCLK cycle.

LinusHeu, 2023-06-21:
> in Phillips (not LJ) mode, as there's an extra BCLK per LRCLK cycle.

There are still16 bits. They're just shifted by one. (If that's what you mean.)
[image]

palmerr23, 2023-06-22:
Earle,

Yes, we need to restrict the PIO dividers to integral values only - which is a good thing to reduce jitter.

> For small n, the jitter introduced by a fractional divider may be unacceptable. However, for larger values, this effect is
> much less apparent. (RP2040 datasheet, p338)

As long as the MCLK divisor is an exact factor of the BCLK one, we're fine, as the PIOs are deterministic. 

> Multiple state machines can be started simultaneously, and the deterministic nature of PIO means they can stay perfectly synchronised. (p312)

Adjusting sys_clk allows pretty close approximations to correct sample rates.  For instance sysclk = 147.6MHz @48k samplerate and MCLK = 192 * BCLK, has a BCLK divisor of 32 and MCLK of 8 for 24 bit samples. That sysclk provides OK divisors for all multiples of 16k sample rates and 192, 384 and 768 MCLK multipliers. 

For 44.1k: sysclk = 135.6 MHz, MCLK = 256 * BCLK. MCLK div = 8 and BCLK = 32.

I've tested almost all of the combinations of sample rate and MCLK multiplier that the PCM5102 supports, and they work OK - i.e. acceptable noise and distortion figures for a  breadboard  hookup. The chart below is 48k sample rate, 768 multiplier.  Others combinations show similar results.

[image]

I haven't yet started validating LJ and 16 bit options, but I don't see too many obstacles. After that, I'll sort out changing parameters on the fly.

Another thing I haven't yet worked out is how (or if) I should try to force the second SM onto the same PIO. 

Also, does duplex (DIN and DOUT together) work? It looks like I need to define two separate I2S instances, which would be OK, though expensive on pins.

earlephilhower, 2023-06-22:
Nice work!

Re changing sys_clk...I'm not sure of the implications there.  I would expect that to cause issues with some of the SDK `alarm`s and other timeouts and things like `millis`.  I guess we could hardcode some values in the menu to complement the nice round ones there already.  That would be much less impactful.

> Another thing I haven't yet worked out is how (or if) I should try to force the second SM onto the same PIO.

Don't think there's any real reason to limit it.  They are all driven from the same clock source.  The `PIOProgram` class **tries** to place everything on the 1st PIO instance first, and only if that fails will it use the 2nd PIO.  So by default you're pretty much guaranteed unless you have 3 other things using PIOs before starting I2S.  In that case the I2S main will be on PIO0.SM3 and the MCLK will be on PIO1.SM0.  But, again, they're all identical.  If we needed to align them we would need to use the `WAIT IRQ` insn.  (If they were on the same PIO there are masked pio_enable` APIs which can start multiple PIOs on the same clock edge).

> Also, does duplex (DIN and DOUT together) work? It looks like I need to define two separate I2S instances, which would be OK, though expensive on pins.

No, they do not.  The PIO code would get rather involved to make that happen and you'd shrink their FIFOs by half.  If there was an easy way to make the PIO driven off of an external clock it might be easily done to take the I2S.out's BCLK.out pin as the I2S.in's BCLK.in pin.  But I am unaware of anything possible like that and you'd need to worry about what edge to latch things...

palmerr23, 2023-06-22:
I'm not having any difficulties with the changed sysclk frequencies and USB. Hardcoding would prevent being able to change sample rates from the 8k set to the 44.1k set on the fly. I'm not sure that would be a real issue, in practice.

I'll create a little NTP synchronised experiment for millis(). It will also validate WiFi and SPI.

I'll take a look at slave mode once I've finished with MCLK (16 bit samples have now been tested and are OK) - which would allow several SMs to source off one clock set (one SM master, several SM slaves).

BTW, I'm wondering why you went to the trouble of creating the _swap modes. Different hardware pin ordering is easily solved at the PCB layer.

I might also take a look at TDM down the track, as I have some CS42448 (8 in 8 out) hardware that works on Teensy 4.x that I could port across. It would interesting to see if the PICO has enough grunt for that. [https://github.com/palmerr23/AudioToy](url)

palmerr23, 2023-06-22:
LJ now tested OK.

LinusHeu, 2023-06-22:
Sorry if this was already answered here but I missed it.
1. What's the formula to calculate optimal sysclk? (Maybe to overclock the pico while still having an optimal sysclk.)
2. The optimal sysclk is only needed when using a MCLK. When using I2S without MCLK, sysclk doesn't have to be adjusted. Right?

I think it would be good to add both answers to the documentation as well.

palmerr23, 2023-06-22:
millis() looks OK.

I ran NTP comparisons over more than an hour at both sysclk frequencies and saw little (really no change).

Clock 147600000

NTP time sync: Thu Jun 22 05:37:05 2023
GMT – millis() = 19906

 NTP time sync: Thu Jun 22 07:47:06 2023
GMT – millis() = 19906, change 0

Clock 135600000

NTP time sync: Thu Jun 22 07:51:14 2023
GMT secs since midnight 28274
GMT – millis() = 28255

NTP time sync: Thu Jun 22 11:16:16 2023
GMT secs since midnight 40576
GMT - millis diff 28255, change 0

palmerr23, 2023-06-22:
Linus,

I simply had to test go by  to achieve the optimal clock. Picking a set of sys_clk parameters and then testing the resulting sysclk as being close to valid integer multiple of the desired MCLK. Once MCLK is satisfied, the BCLK divisor is fairly straightforward to calculate.

All the formulae are in the RP2040 Datasheet. I've attached a document that I wrote while working out the logic.  [PICO AUDIO I2S.docx](https://github.com/earlephilhower/arduino-pico/files/11833361/PICO.AUDIO.I2S.docx)

LinusHeu, 2023-06-22:
Thank you. Very interesting and more complicated than I thought :D

DatanoiseTV, 2023-06-22:
I have access to a DSLogic LA, which can sample at up to 400Mhz, if that is any help.

DatanoiseTV, 2023-06-22:
> I'm not having any difficulties with the changed sysclk frequencies and USB. Hardcoding would prevent being able to change sample rates from the 8k set to the 44.1k set on the fly. I'm not sure that [...]

+1

jfrey-xx, 2023-06-22:
Hello,

Not sure if this will help or not, on my end for determining a CPU freq that is a nice multiple of the MCLK I've been using `vcocalc.py` from the pico SDK (e.g. I've been using 30khz x 256  with 153.6Mhz SGTL5000, using for MCLK the code I posted above).

palmerr23, 2023-06-23:
@Jfrey,

Sorry I didn't see your code before I started, I've ended up with something quite similar - particularly the PIO code, which was a no-brainer.

Yes vcocalc.py works fine. You can also call check_sys_clock_khz() to see if a sysclk is legal. You could write a nice little program that tests all the permutations - there are four variables for sysclk and a further two for PIO dividers. Quite a few iterations would be required!

I've restricted the PIO dividers to 16:0 integer values to avoid jitter (Datasheet p 338).  I've also been less restrictive on the MCLK multiplier - simply restricting it to multiples of 64, so that a wider range of devices can be accommodated.

@Datanoise - thanks for the offer - I'd be very happy for you to test the code! However my comment about the poor MCLK alignment in the analyzer images was mainly about visible 'jitter'. The audio test results suggest that that's under control and my frequency meter validated all the frequencies.

DatanoiseTV, 2023-06-23:
@palmerr23 I can check it on monday if you send me some code. Cheers.

commit cc5d1779 (2023-07-07): Add MCLK support for I2S, optimize clocks for jitter-free playback (#1555)

Fixes #1065
closed 2023-07-07 by earlephilhower