earlephilhower/arduino-pico #1066: Feature request: Add TDM mode for I2S

DatanoiseTV, 2022-12-24:
Codecs (with more than 2 channels use TDM (Time Domain Multiplex) to provide up to 8 Channels of Inputs or Outputs per I2S port.
For instance, the CS42448 Codec has 6 Inputs and 8 Inputs which can be used using one single I2S interface.

Adding TDM support would add a big range of new possibilities.

earlephilhower, 2022-12-24:
If this needs no add'l signaling, it may be doable right now once the base I2S setup has all the kinks worked out.

The PIO code gets passed in the # of bits per phase of WLCLK, so just sending in 6x that number and then interleaving the 6 channels in the memory buffer would "just work."

DatanoiseTV, 2022-12-25:
It doesn't need any extra signaling.

[image]

earlephilhower, 2022-12-26:
Ah, that's almost trivial to do.  Change the fPIO from `2 * bps * hz * 2` to `channels * bps * hz  * 2` and I think everything else will "just work."  Tell the PIO program to send `n * bps - 2` not `2 * bps - 2`.  Some API for TDM channel config and writing needs to be there (I'd think we would want to require you to write all TDM samples in series, and not support writing all ch1 then all ch2 then all ch3...  simply because it's a lot of memory and wasteful CPU-based interleaving).

PR #1673 (2023-08-30): Add TDM support to I2S
Fixes #1066

Implements a simple TDM mode for the I2S output object.
closed 2023-08-30 by earlephilhower