earlephilhower/arduino-pico #1067: GPIO performance

neilger, 2022-12-26:
Thank you @earlephilhower for the great work on this.

Here is a performance test of GPIO-core timing that connects two pins, writes a bit out, reads it back in, and inverts it:

https://pub.pages.cba.mit.edu/ring/

9.17 MHz (Arduino, SIO)  
0.58 MHz (Arduino, digitalRead/Write)  
0.039 MHz (MicroPython)  

Have you considered adding digitalReadFast/WriteFast alternatives like e.g. https://github.com/SpenceKonde/megaTinyCore for AVRs, for a more portable option for use cases that don't fit on PIOs?

earlephilhower, 2022-12-26:
Thanks the the info.  As this is something of a one man show I have to prioritize functionality over performance . :)

@neilger can you post the test INO file?  it's literally a `#define` to make the `digitalReadFast`/`digitalWriteFast` renaming the build-in SDK macros.

neilger, 2022-12-26:
@earlephilhower here's the INO version, 12.9 MHz at 250 MHz: https://pub.pages.cba.mit.edu/ring/GPIO/RP2040/ring.RP2040.ino

Is the #define built-in, or were you suggesting I add that to my code?

FYI, the MicroPython version can be sped up ~10x to 0.35 MHz with a native function decorator: https://pub.pages.cba.mit.edu/ring/GPIO/RP2040/ring.RP2040.py

earlephilhower, 2022-12-26:
Look at #1069.  You can use the PR or just copy-n-paste the macros in your code to test.  It's be the about same as the SIO access ones.

Also, try `-O2` in Tools->Optimization and re-run.  You should get a boost there, too, I would think.

neilger, 2022-12-27:
Here's what the optimizations do, using sio_hw:

- 250 MHz
    - -Os 12.9 Mhz
    - -O 10.7 MHz
    - -O2 10.7 MHz
    - -O3 9.8 MHz
    - -Ofast 9.6 Mhz
- 133 MHz
    - -Os 9.17 MHz
    - -O 6.88 MHz
    - -O2 7.60 MHz
    - -O3 7.67 MHz
    - -OFast 7.67 MHz

Using instead:

#define digitalWriteFast(pin,val) gpio_put(pin,val ? 1 : 0)  
#define digitalReadFast(pin) (gpio_get(pin) ? 1 : 0)  

at 250 Mhz rings at 8.77 MHz, although with more timing variability:

[image]

than directly calling sio_hw:

[image]

earlephilhower, 2022-12-27:
Nice analysis, thanks!  I wonder what could be causing jitter in the `#define` version, very odd.  Also interesting to see `-Os` (default, optimize for size) is the best performing.  I suppose that has to do with the onboard XIP caches.

Let me see if I can use the `sio->` version or tweak the way I'm using the SDK calls...

earlephilhower, 2022-12-28:
My logic analyzer doesn't have enough BW to tell me what the frequency of the latest push is, only that it's between 8 and 10 MHz (core at 133 default), so I think we're good.   Basically it implements your macros using the SDK calls.

closed 2022-12-28 by earlephilhower

neilger, 2022-12-29:
@earlephilhower with:

#define digitalWriteFast(pin,val) (val ? gpio_set_mask(1 << pin) : gpio_clr_mask(1 << pin))  
#define digitalReadFast(pin) gpio_get(pin)  

I get 8.76 MHz ring (at 250 MHz clock); eliminating the function calls:

#define digitalWriteFast(pin,val) (val ? sio_hw->gpio_set = (1 << pin) : sio_hw->gpio_clr = (1 << pin))
#define digitalReadFast(pin) ((1 << pin) & sio_hw->gpio_in)

I get 12.9 MHz (matching reading/writing to sio_hw directly).

earlephilhower, 2022-12-29:
That's very weird!  The "function call" should be an inline anyway and fall into the exact same generated code:
https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/hardware_gpio/include/hardware/gpio.h#L697-L699

earlephilhower, 2022-12-29:
@neilger would you like to do a PR with the hand-instantiated `sio_hw` calls?

neilger, 2022-12-30:
@earlephilhower: https://github.com/earlephilhower/arduino-pico/pull/1077

PR #1069 (2022-12-28): Add digitalWrite/ReadFast macros
Fix #1067
commit 47b18ea8 (2022-12-28): Add digitalWrite/ReadFast macros (#1069)

* Add digitalWrite/ReadFast macros

Fix #1067

* Update using @neilger's macros