earlephilhower/arduino-pico #1068: Linux upload mountpoint

nottledim, 2022-12-26:
Some discussion of this is in #420.  The problem is arduino fails to upload the firmware to the rpi.  The program uf2conv.py is what's doing the work.  It assumes various resources exist.

Specifically a directory like /media/$USER has to exist.  This doesn't necessarily create itself and the error emitted when it doesn't exist is the obscure "No drive to deploy". You need to be root to create that directory so arduino ide can't create it if it's not there. The documentation doesn't mention this requirement.

But why create it there? Could it not be created in /tmp/arduino_build_nnn which is specific to that session and owned by $USER. Also because it's all ephemeral the name is arbitrary and it can be created on the fly. I guess you'd have to modify uf2conv.py to accept an argument specifying the rootpath for the mountpoint.

This is a real time-waster issue - please, at least update the documentation to mention the requirement.

earlephilhower, 2022-12-26:
The directory isn't created by the tool, it's part of your distro's infrastructure.  Users normally can't even mount filesystems in any case, in my experience.

The script now checks a series of "well known" spots on different distros (RH, Ubuntu, Arch, maybe others) and if all those spots fail it will abort with the error (direct from Microsoft, so to change it I'd suggest going upstream to their UF2 repo).

What distro are you using, and where exactly is the automout spot for USB media?  It's trivial to add another spot to check to the tool...

nottledim, 2022-12-27:
I have a Debian system. I have no idea how auto mounting of USB devices works these days.  Over the years it's changed and it's not something I use very often.  Dmesg has this line which explains a lot:

`
 udisksd[1739]: Mounted /dev/sdf1 at /media/dick/RPI-RP2 on behalf of uid 1005
`

another new-fangled thing that's found it's way onto my system :-(  As ever the Archlinux wiki is useful: https://wiki.archlinux.org/title/udisks

Your discussion https://github.com/earlephilhower/arduino-pico/discussions/913 has list of things to check.  I didn't appreciate the implications of "a list of filesystems".

Oh well live and learn.  Thanks for your help.

closed 2022-12-27 by nottledim

AnachronisticPenguin, 2023-01-08:
If your Arduino IDE is installed by Flatpak you can resolve this error by configuring Flatpak to give the application access to the host file system. I had the same error mentioned in this issue on Pop!_OS 22.10 and this command resolved it for Arduino IDE v2:

`flatpak override --user --filesystem=host cc.arduino.IDE2`

Older IDE versions (e.g., 1.8.x or older) have a different application name:

`flatpak override --user --filesystem=host cc.arduino.arduinoide`