earlephilhower/arduino-pico #1082: Including WiFi support breaks an already working sketch

lu7did, 2022-12-31:
I'm working on a sketch which collect samples of a signal and process them after some pre-defined time, the processing is somewhat complex but it works Ok. Adding Wifi support to be able to sync the clock using NTP make the operation unstable, it even hang after some 45 seconds or so.

The ADC collection segment follows, setup_adc() is called once whilst collect_adc() is called every 160 mSecs or so.
```
void setup_adc() {
  adc_gpio_init(ADC0 + ADC_CHANNEL);
  adc_init();
  adc_select_input(ADC_CHANNEL);
  adc_fifo_setup(
    true,    // Write each completed conversion to the sample FIFO
    true,    // Enable DMA data request (DREQ)
    1,       // DREQ (and IRQ) asserted when at least 1 sample present
    false,   // We won't see the ERR bit because of 8 bit reads; disable.
    false     // Shift each sample to 8 bits when pushing to FIFO
  );
  adc_set_clkdiv(CLOCK_DIV);
  sleep_ms(200);
  dma_chan = dma_claim_unused_channel(true);
  cfg = dma_channel_get_default_config(dma_chan);
  channel_config_set_transfer_data_size(&cfg, DMA_SIZE_16);
  channel_config_set_read_increment(&cfg, false);
  channel_config_set_write_increment(&cfg, true);
  channel_config_set_dreq(&cfg, DREQ_ADC);

}
void collect_adc() {

  adc_fifo_drain();
  adc_irq_set_enabled(false);
  adc_run(false);
      
  dma_channel_configure(dma_chan, &cfg,
      fresh_signal,    // dst
      &adc_hw->fifo,  // src
      CAPTURE_DEPTH,          // transfer count
      true            // start immediately
      );
  adc_irq_set_enabled(true);
  adc_run(true);
}
```
The processing of the samples is complex but straightforward codewise
```
    collect_adc();
    tstart = time_us_32();
    while (true) {
      if (flag_first) {
        dma_channel_wait_for_finish_blocking(dma_chan);
        flag_first = false;
        break;
      }
        for (int i = 0; i < nfft; i++) {
            old_signal[i] -= DC_BIAS;
        }
        adc_error = inc_extract_power(dma_chan, old_signal,false);
        if (adc_error != 0x00) {
           break;
        }
       while (dma_channel_is_busy(dma_chan));
       break;
    } //true, pseudo infinite loop
  } //whole reception cycle
```
As said the above code works ok.

Then I added the following lines
```
int setup_wifi() {
  WiFi.begin("MyISPSSID", "mipassword");
  delay(1000);

  uint32_t t = time_us_32();
  
  while (WiFi.status() != WL_CONNECTED) {
    if (time_us_32() - t > wifi_tout * 1000000) {
      return WiFi.status();
    }
  }
  char ipstr[16];
  sprintf(ipstr,"%s",IpAddress2String(WiFi.localIP()).c_str());
  delay(1000);
  return 0;
}
```
The sketch works, the connection is made to the WiFi AP succesfully, the data collection cycle operates for some 30 secs
correctly and then the whole sketch hangs.

Probably the above isn't enough to profile the possible reason, but I would like to understand if the WiFi implementation has some interaction or dependency with the ADC operation, or the DMA reading of ADC samples or requires some periodic
call somewhere to operate or some interrupt handling jeopardizes the working.

Thanks in advance, Pedro

earlephilhower, 2022-12-31:
The snippet really isn't enough to go on, but remember that the Pico W steals several pins for the WiFi chip PIO-based SPI interface.  Maybe you're stepping on one of those pins.

If you can't make a small snippet (MCVE) that reproduces the problem, your next best bet is to hook up a PicoProbe and get a GDB stack dump to see where things are when it hangs (i.e. is it a panic() or an infinite loop somewhere, etc.)

earlephilhower, 2022-12-31:
Oh yeah, the DMA may also be an issue.  I seem to remember there being a bug open in the Pico-SDK repo about the WiFii code assuming/stomping on certain hardcoded DMA and IRQs.  Check there if you have a chance...

lu7did, 2022-12-31:
Well, I'll put together a snippet which is essentially the code segments I put with some delays to represent the processing taking place, if you can guide me on what you need to see to have a better grasp of what is going on I will put it together.
But one advance, if I call WiFi.disconnect() after connecting to the AP and checking the NTP server for time synchronization the program get back to work and no hang occurs. so it's something related to either the ADC or DMA or both.

earlephilhower, 2022-12-31:
The snippet really need to be complete and self contained and small (i.e. complete app that anyone can run it and reproduce the failure).  OTW it's not really possible to look at other than 1st-level eyeballing...

FeuerSturm, 2023-01-02:
@lu7did 
I do have a similar yet different issue, I developed a sketch with v2.3.0 and it works fine from that version to 2.4.1, the later versions just hang the sketch after round the same 30sec.
I haven't had time to investigate further, but downgrading to 2.4.1 and trying your sketch won't hurt.

earlephilhower, 2023-01-04:
No update in a few days so closing until a complete MCVE is available, but I think this is just related to the PicoW pinous.

Per the Pico W datasheet, the following GPIOs are used by the WiFi chip interface:
````
A few RP2040 GPIO pins are used for internal board functions:
* GPIO29 OP/IP wireless SPI CLK/ADC mode (ADC3) to measure VSYS/3
* GPIO25 OP wireless SPI CS - when high also enables GPIO29 ADC pin to read VSYS
* GPIO24 OP/IP wireless SPI data/IRQ
* GPIO23 OP wireless power on signal
````

Since you're using the ADC, you may be stomping on the ADC3 channel(GP29) or other IOs or ADC registers that are used by the WiFi interface.  At that point, all bets are off and you may hang quickly or slowly or just lose WiFi...

closed 2023-01-04 by earlephilhower