earlephilhower/arduino-pico #1085: attachInterrupt to GPIO pin does not work in constructor of global object

per-magnusson, 2023-01-01:
First: Thanks for your efforts to provide support for RP2040 in the Arduino IDE!

Short description: 

This code:

`attachInterrupt(pin_a, isr_a, CHANGE);`

has no effect if it is run in the constructor of a global object. It works fine however if it is moved to setup().

Longer description:

I am new at trying to use Rapsberry Pi Pico in the Arduino IDE (have mostly been using Teensys before) and when I tried to reuse an interrupt based rotary encoder class I wrote a few years ago, which works fine with e.g. Teensy 4, I ran into problems. The ISRs never got called. After some troubleshooting, I discovered that if I moved the attachInterrupt calls so that they were not called in the global constructor (i.e. before setup()) to inside setup(), things started working. 

So it seems like some magic setup code is run between the constructors of global objects and setup() that either cancels the effect of the attachInterrupt calls, or does something that needs to be done to make later attachInterrupt calls have an effect.

Is it possible to make some change that would allow attachInterrupt to be functioning even if it is used in a constructor of a global object, and thus called before setup() is run?

I can provide example code that demonstrates the issue if that would be of help, but perhaps the above description is enough to pinpoint the issue.

maxgerhardt, 2023-01-01:
Calling Arduino APIs from constructors is like playing russian roulette with 5 of 6 chambers loaded. See: ["static initialization order fiasco"](https://isocpp.org/wiki/faq/ctors#static-init-order). You in general don't have a guarantee whether your constructor will run before or after a certain other object's constructor code. 

In regards to attachInterrupt(), you will see that it's writing the pin and callback function you attach on is written into a std::map object

https://github.com/earlephilhower/arduino-pico/blob/a7cf5cd1caf1afc8aebc8723776847227fed8966/cores/rp2040/wiring_private.cpp#L114-L119

https://github.com/earlephilhower/arduino-pico/blob/a7cf5cd1caf1afc8aebc8723776847227fed8966/cores/rp2040/wiring_private.cpp#L75

But you could have the situation where your constructor code runs before the std::map _map one's, puts something in there, then the std::map constructor runs and resets some variables back to 0, seemingly loosing the content of map.

Classes whose constructors only copy their arguments into member variables and then have a separate `.begin()` function are much safer and do not depend on the minute details of C++ constructor call order.

I'll have to confirm that in the debugger to be 100% sure that that's what's happening (and not something in Pico-SDK), but that should be the most probable explanation of your problem.

per-magnusson, 2023-01-01:
Thanks for enlightening me. I was not aware of the "static initialization order fiasco", but the explanation does of course make sense. I will rewrite my code as you suggest to just store the relevant information in the constructor and then require a .begin() to actually call functions like attachInterrupt(). Even though my original code has been working fine in other projects (mostly with Teensys), there are obviously latent bugs that need to be fixed there.

Again thanks for your helpful explanation.

closed 2023-01-01 by per-magnusson