earlephilhower/arduino-pico #1086: Add cli() and sei() as aliases for noInterrupts() and interrupts()

per-magnusson, 2023-01-01:
cli() and sei() are used in lots of Arduino code. As I understand it, they are synonyms for noInterrupts() and interrupts() as discussed here:

[https://forum.arduino.cc/t/interrupts-cli-sei/127779](https://forum.arduino.cc/t/interrupts-cli-sei/127779)

However, I had to change some legacy code from cli() to noInterrupts() etc to make it work with arduino-pico, so cli() and sei() do not seem to be defined in this environment.

To improve portability of old code, it would be good if these functions were added also for the RP2040. It might be sufficient to add the following to arduino.h:

#define sei() interrupts() 
#define cli() noInterrupts()

earlephilhower, 2023-01-01:
@per-magnusson, why not do a PR?  Add it after the `digitalWriteFast/readFast` macros in `Arduino.h` and it should be all set.

per-magnusson, 2023-01-01:
Thanks for your reply. I did not create a PR only because I am a total noob when it comes to git (I am much more comfortable with SVN...) Anyway, I have now tried to create a PR with this change. Not sure I succeeded, but if not, let me know and I will make further attempts to figure it out. Sorry for not being up to speed on git and github.

earlephilhower, 2023-01-01:
Thanks for the PR, no problems with it!  You're giving me flashbacks with `svn`.  At least it's not `cvs`... :sob: 

This summarizes most users' first few times with `git`.  https://xkcd.com/1597/

It does get better, though!

closed 2023-01-01 by earlephilhower

per-magnusson, 2023-01-02:
You are the one who deserves thanks. Learning git (better than in the xkcd strip...) is on my todo list, although not necessarily on the top. Being "forced" to do something with git here and there is probably helpful in getting there eventually.