earlephilhower/arduino-pico #1094: Japanese or LSBJ I2S format

tomcombriat, 2023-01-03:
Hello!
I have been fiddling a bit with the I2S library of arduino-pico (thanks for it!) and a PT8211 DAC. I encountered some weird behavior that I think are due to the fact that the PT8211 is not really following the I2S data format but the Japanese variant (also called LSBJ - Least significant bit justified). The main difference is that, for the LSBJ the MSB is sent right after the change of WS (Word Select) whereas in I2S the LSB is sent there (quickly followed by the next MSB).

Is there a way to get this other format working cleanly with the pico I2S that I did not see? Or is fiddling a bit with the bit (pun intended) ordering before sending to `I2S::write()` needed?

Thanks!

earlephilhower, 2023-01-03:
To get that working I think you really want a different PIO program.  It shouldn't be complicated (and should be smaller than the I2S one) , and you'd probably pass in (# of bits), not (# of bits - 2).  The rest of the logic would be the same.

Alternatively since it's just effectively shifting the signal 1 bit in advance, you could (as an ugly hack) adjust the I2S::write to handle the shifting.  It would need to buffer the filled 32-bit value until the next write, where it would shift-left the bits and set bit 0 to the MSB of the next write.  It seems like it would be easier to just write a new PIO program than this, though.

tomcombriat, 2023-01-03:
Hi,
Thanks for your answer!
I was thinking about doing the ugly hack but I found out this would be over-complicated because of the need to buffer the value and also to deal properly with the sign bit upon shifting.
I'll try to commit myself to make a different PIO program. Just a quick question before hand, would you be interested in a PR if I get that to work? If so, should it be part of a different library (for eg <LSBJ>) or be part of this one with a possibility to switch between the two formats upon construction?

Best,

earlephilhower, 2023-01-03:
@tomcombriat can you give #1095 a try and report back?  It was literally about 5 mins including hooking up the LA...

tomcombriat, 2023-01-03:
Hi, I will! I was just opening the pio file actually. (I cannot guarantee full testing today unfortunately)

Thanks!

earlephilhower, 2023-01-03:
No worries.  Just add a `i2s.setLSBJFormat()` call before `i2s.begin()` and it should run in LSBJ mode.  So if you have something already written (or even SimpleTone) you can add the call and re-run to see immediate effect.

tomcombriat, 2023-01-03:
Works like a charm from initial test with one nitpick: left and right channels are inverted (that is also a difference in the LSBJ format).

Just out of curiosity, trying to understand your changes (bare than I am not familiar with PIO programming), I could not see traces of the 
> you'd probably pass in (# of bits), not (# of bits - 2).
you was referring too. I am not able to test if these two bits are remained as my test board is too noisy to get them out.

Thank you very much!

earlephilhower, 2023-01-03:
> Just out of curiosity, trying to understand your changes (bare than I am not familiar with PIO programming), I could not see traces of the (#) vs (#-2)

It turned out was simpler just to keep the I2S PIO logic but delay the LRCLK one BCLK.  So the only program difference it the LRCLK changes moved down a couple lines.  Simple and easy enough to understand is good enough for now.  I still think you could probably reduce the # of PIO insns but it's not worth the effort IMHO.

earlephilhower, 2023-01-03:
(To be clear, it is shifting out all 16 bits.  The bitcnt is reduced by 2 as sent in because there's a couple implied bit shifts before the decrement anyway.  All 16/24/32 bits are sent out in I2S or LSBJ mode...)

closed 2023-01-03 by earlephilhower

tomcombriat, 2023-01-03:
Thank you very much for both this modification and taking the time to explain!
Best,

PR #1095 (2023-01-03): Add LSBJ format support for I2S
Fixes #1094
commit 6fe6c474 (2023-01-03): Add LSBJ format support for I2S (#1095)

Fixes #1094