earlephilhower/arduino-pico #1102: usb serial

natxopedreira, 2023-01-06:
Hello!!

I'm using this framework for a time, and is great, thanks for this one!!

This is mainly a question about how this works as i would like to solve a issue i'm having.

Sometimes pico don't show as a com port on system, i dig a bit and seems is a "issue" with pico sdk and normal recommendation is to set a delay after "stdio_init_all" to give it a bit of time.

Some examples

[https://forums.raspberrypi.com/viewtopic.php?t=300136](https://forums.raspberrypi.com/viewtopic.php?t=300136)
[https://forums.raspberrypi.com/viewtopic.php?t=308068](https://forums.raspberrypi.com/viewtopic.php?t=308068)

It only happens on rare occasions and i was not able to find the root of what it cause.

What would be the best way to implement "that wait a bit for usb/serial before continue"??

In my case i dont allways have the usb connected so cant put a totally blocking wait
```
stdio_init_all();

// wait 10 segs max for usb
uint32_t t_out = millis();
while(!tud_cdc_connected() && millis()-out < 10000)
{
 sleep_ms(100);
}

usb_serial.begin(115200);
```

And they just put the wait after the `stdio_init_all() ` call, but in the ardu way there no this call.

What will be the equivalent method using here instead this stdio_init_all() ??

Regards

earlephilhower, 2023-01-06:
You could use 
````
Serial.begin();
while (!Serial && millis() < 10000) { delay(1); }
````
(`operator bool Serial` returns the connected state).

Normally I just throw in a `delay(5000)` at the start of the sketch.  The problem is even after it reports connected, the OS (in my experience) still takes some time to recognize things and start capturing bits. so the prior code may still skip some chars.

closed 2023-01-06 by earlephilhower

natxopedreira, 2023-01-06:
Ah ok, i see.

In my case i dont mind skipping characters, problem is that device don't shows on OS, will try setting that delay

Thanks