earlephilhower/arduino-pico #1109: Support for chinese YD-2040 boards?

prosper00, 2023-01-11:
Any appetite to add support for the inexpensive Chinese boards that came on the market a year or so ago? 
See https://www.aliexpress.com/item/1005004004120604.html

Documentation (schematics, BOM, pinout) (provided by the seller) attached:
[YD-2040-2022-V1.1-SCH.pdf](https://github.com/earlephilhower/arduino-pico/files/10396038/YD-2040-2022-V1.1-SCH.pdf)
[image]
[image]
[YD-RP2040-Inch-SIZE.pdf](https://github.com/earlephilhower/arduino-pico/files/10396039/YD-RP2040-Inch-SIZE.pdf)
[image]
[YD-RP2040-Metric-SIZE.pdf](https://github.com/earlephilhower/arduino-pico/files/10396040/YD-RP2040-Metric-SIZE.pdf)

earlephilhower, 2023-01-12:
That's very simple to do, but without having some to test I'm not sure what the actual flash should be in the board definition.  The BOM is one thing, but do we really know what is actually being stuffed on the board?

Do you have some in hand to test?  If so, could you try the `Generic` board and go thru the different flash options (i.e. load blink.ino, set boot stage 2 to A/4, upload and run, set boot stage 2 to A/2, upload and run, etc.) . Some will work, others will fail to boot  I'm thinking the `W25Q64JV /2` or `/4` will be the right one given the BOM.

prosper00, 2023-01-12:
I don't have an option for W25Q64JV /2 - but W25Q16JVxQ /4 and W25Q64JV /4, and 'Generic SPI /2 and /4' options all work for the 4MB and 16MB models.

closed 2023-01-14 by earlephilhower

tistructor, 2023-01-19:
The files and the "variant" folder for this board have not been created

earlephilhower, 2023-01-19:
Yeah, sorry.  Just added!

commit 089cd442 (2023-01-14): Add VCC-GND YD-RP2040 board

Fixes #1109
PR #1120 (2023-01-14): Add VCC-GND YD-RP2040 board
Fixes #1109
commit 1328f780 (2023-01-14): Add VCC-GND YD-RP2040 board (#1120)

Fixes #1109