earlephilhower/arduino-pico #1110: usage of UART1 with SDK funktionality, ist blockt by"Arduino Serial."

kendo55, 2023-01-12:
Problem: 
with option TEST_NR == 2
the use of Serial ist blocking the output on UART1 until we open the serial monitor in Arduino 1.8.9. and 2.7.0 arduino-pico.
That means, without starting the serial monitor, the UART1 output is missing.

with option TEST_NR != 2 the output on UART1  runs.

void init_uart1_test1() is a codesample from the SDK

````
/*
TEST UART1
-------------
usage of the UART1: first without RX/TX IRQ  ... later with
*/
#include "pico/stdlib.h"
#include "hardware/uart.h"
#include "hardware/irq.h"
#include "hardware/clocks.h"

#define TEST_NR   1

void setup() {
//--------------------
  #if (TEST_NR == 2)
    //Serial.begin(115200);
    //while (!Serial) {}  // wait for serial port to connect. Needed for native USB
  #endif

  init_uart1_test1();

}

void loop() {
//----------------------------

delay(2000);
uart_puts(uart1, "print over UART 1 ........\n");
Serial.println("print over Serial.println ........");

}

void init_uart1_test1(){
//------------------------------------------  

#define UART_ID uart1
#define BAUD_RATE 115200
#define DATA_BITS 8
#define STOP_BITS 1
#define PARITY    UART_PARITY_NONE

// We are using pins 0 and 1, but see the GPIO function select table in the
// datasheet for information on which other pins can be used.
#define UART_TX_PIN 8   // PICO-Pin 11
#define UART_RX_PIN 9   // PICO-Pin 12

// Set up our UART with a basic baud rate.
    uart_init(UART_ID, 2400);

    // Set the TX and RX pins by using the function select on the GPIO
    // Set datasheet for more information on function select
    gpio_set_function(UART_TX_PIN, GPIO_FUNC_UART);
    gpio_set_function(UART_RX_PIN, GPIO_FUNC_UART);

    // Actually, we want a different speed
    // The call will return the actual baud rate selected, which will be as close as
    // possible to that requested
    int actualBR = uart_set_baudrate(UART_ID, BAUD_RATE);

    // Set UART flow control CTS/RTS, we don't want these, so turn them off
    uart_set_hw_flow(UART_ID, false, false);

    // Set our data format
    uart_set_format(UART_ID, DATA_BITS, STOP_BITS, PARITY);

    // if we want to do this character by character, then Turn off FIFO's - 
    // now they are turned on.
    uart_set_fifo_enabled(UART_ID, true); // true  false  offenhs
 
/*
    // Set up a RX interrupt
    // We need to set up the handler first
    // Select correct interrupt for the UART we are using
    int UART_IRQ = UART_ID == uart0 ? UART0_IRQ : UART1_IRQ;

    // And set up and enable the interrupt handlers
    irq_set_exclusive_handler(UART_IRQ, on_uart_rx);
    irq_set_enabled(UART_IRQ, true);

    // Now enable the UART to send interrupts - RX only
    uart_set_irq_enables(UART_ID, true, false);
*/
  
     uart_puts(UART_ID, "\n>>> Init UART 1 o.k.\n");
}
````

earlephilhower, 2023-01-12:
Can you be more specific about what the problem is?  Your code explicitly says to wait for Serial connectivity before continuing, so the explanation doesn't make sense.  Drop the `while (!Serial) {}` and it will fall through and run w/ or w/o USB connection.

kendo55, 2023-01-12:
Version1:
=======
If I  set #define TEST_NR   1 --> ( !=2)  ---> Serial.begin   NOT set:

on UART1 the Output is:
-------------------------------
```` 
>>> Init UART 1 o.k.
print over UART 1 ........ 
etc.
```` 
on USB/COM the Output is:
--------------------------------
``
print over Serial.println ........
etc.
``
``
     ---->was not to be expected because the start with Serial.begin is missing
        **---> Question 1:**  why?

Version2:
=======
If I  set #define TEST_NR  2 --> (TEST_NR==2)  --->Serial.begin   set:

the output only starts when the USB-Serial Monitor is on (the port is opened) !

Now that's clear for me, because on USB-Serial the definition of  "if(Serial)" is different (sorry)

BUT the **Question 1** above stays.

AND 
one addidional Question: 
Does all SDK-Funkionality (Rx/Tx IRQ) of the UART1 runs, even we have initiate:   a) Serial.beginn()  and/or b) Serial1.biginn() 

[Arduino doc](https://www.arduino.cc/reference/en/language/functions/communication/serial/ifserial/)

if(Serial)

a)  Indicates if the specified Serial port is ready.

b) On the boards with native USB, if (Serial) (or if(SerialUSB) on the Due) indicates whether or not the USB CDC serial connection is open. For all other boards, and the non-USB CDC ports, this will always return true.

earlephilhower, 2023-01-12:
If the problem is the USB serial port works even w/o a `Serial.begin()`, that is expected because we need the USB serial port to do uploads so it is always begun.  (The IDE sends a special 1200bps DTR toggle sequence which is "standard" on the Arduino side which causes a reboot to the UF2 ROM).

You can also run all 3 serial ports, `Serial`, `Serial1`(UART0) and `Serial2` (UART1) in parallel at the same time, that's normal behavior as well.

However, `Serial1` and `Serial2` do require a `begin(baud)` because they need to know what speed to operate at.  W/O calling `Serial1/2.begin(bps)` no UART config is touched (no registers set, no IRQs hooked, etc.).  For USB the baud is ignored, of course.

closed 2023-01-12 by earlephilhower

kendo55, 2023-01-12:
Thank you very much!
Last question: How could we turn off the USB serial features like in real SDK development?: pico_enable_stdio_usb(${OUTPUT_NAME} 0)

I have the suspicion that the USB part could lead to blockages.

earlephilhower, 2023-01-12:
`Tools->USB Stack->No USB`.  You'll need to manually reset to UF2 mode to upload at that point, but no USB will be run.