earlephilhower/arduino-pico #1111: Library description for Arduino IDE

sstaub, 2023-01-12:
The board description is nearly unreadable in the Arduino IDE 2.x. It is too long so you need endless scrolling to see other boards.
[image]

Phoenix1747, 2023-01-12:
Most importantly this makes it impossible to click the "Install" button, because it will be pushed out of the window. But this is more of a general problem for the Arduino IDE, I think. Just writing for completeness.

sstaub, 2023-01-12:
@Phoenix1747 I seen this also on WIN10 when look to updateable boards and the pico is the last item.

maxgerhardt, 2023-01-12:
It seems to be ignoring the `boards` array of the package index json though, as seen [here](https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json) that only declares
```json
          "boards": [
            {
              "name": "Raspberry Pi Pico"
            },
            {
              "name": "Raspberry Pi Pico W"
            },
            {
              "name": "Adafruit Feather RP2040"
            },
```
Aka not with every combination of `(pico-debug)`, `(Picoprobe)` and without it. Seems to be reading it straight from the `boards.txt`.

But if that's the case, then that cannot be made smaller without throwing board definitions out or restructuring it.

commit cc685d53 (2023-01-13): Separate out upload method, add picotool upload

Instead of listing each board three times (normal upload, picoprobe,
and pico-debug uploads), list each board once and use a menu to select
the actual upload method.

Also add in picotool as an upload method for those folks who have trouble
with automounting.

Fix #1111
PR #1112 (2023-01-13): Redo boards menu, separate out upload method, add picotool upload
Instead of listing each board three times (normal upload, picoprobe, and pico-debug uploads), list each board once and use a menu to select the actual upload method.

Fix #1111
commit ae386d43 (2023-01-13): Redo boards menu, separate out upload method, add picotool upload (#1112)

Instead of listing each board three times (normal upload, picoprobe,
and pico-debug uploads), list each board once and use a menu to select
the actual upload method.

Also add in picotool as an upload method for those folks who have trouble
with automounting.

Fix #1111
closed 2023-01-13 by earlephilhower