earlephilhower/arduino-pico #1113: Pico W fails to connect to open WiFi network

jhmaloney, 2023-01-13:
This is a CYW43 driver issue. The same problem cropped up in MicroPython and they apparently found a patch. See [Issue 9016](https://github.com/micropython/micropython/issues/9016)

earlephilhower, 2023-01-14:
Thanks for the info.  Since it's in a submodule of a submodule here, I'd like to sit on this for a bit and see if the Pico SDK 1.4.1 takes up this new patch.  If not, we can revisit this and see about moving up unilaterally...

jhmaloney, 2023-01-14:
Yes, that makes sense. Just wanted to to know about the issue.

Fortunately, there are not too many open networks around -- although our public library's WiFi network is one.

earlephilhower, 2023-02-11:
SDK 1.5 should have the fix.  Need to see if we require the change done to upython:

https://github.com/jcernato/micropython/commit/d8ae1372905f28507992840ebf2218a92babfe16

jhmaloney, 2023-02-11:
Great! Thanks for tracking this issue.

PR #1167 (2023-02-28): Update to SDK 1.5, add alpha-level BT support, use Pico-SDK CYW43 infrastructure
Fixes #1212
Fixes #1113 
Fixes #954 
Fixes #906
closed 2023-02-28 by earlephilhower