earlephilhower/arduino-pico #1118: Problem with Serial Ports

k7ilo, 2023-01-14:
Not really an issue but I cant seem to find the proper information about your arduino-pico core pertaining to Serial port use on Google or anywhere is so I am using this as a last resort.  I keep seeing that your core has pre-defined pins for UART1 and 2 but there is so many conflicting documents out there.  How do you properly use the Serial ports, especially if I need to use i2c also?  Is there a set of documents provided by you or someone who has been using your arudino-pico core for some time?

Thank You

Phoenix1747, 2023-01-14:
Did you read the [arduino-pico](https://github.com/earlephilhower/arduino-pico) documentation? Serial1 is UART0, and Serial2 is UART1. You can change RX and TX pins on both UART interfaces. Please have a look at https://arduino-pico.readthedocs.io/en/latest/serial.html
To avoid conflicts, just set your I2C pins and UART pins accordingly, no problem.

You can see the different UART TX and RX pin pairs here:

[image]

https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf

Hope I could help you a bit.

earlephilhower, 2023-01-14:
@Phoenix1747 thanks, that sums things up pretty simply.  The picture and the docs explain how to config any legal setup of SPI/UART/etc.  Good luck!

closed 2023-01-14 by earlephilhower