earlephilhower/arduino-pico #1123: Arduino IDE cant find Pico in COM Port (Windows)

dev-000, 2023-01-16:
I am having an issue with an RPi Pico board. I have installed the latest Arduino IDE 2.0.3, the Latest Arduino-pico core 2.7.1
The issue is I am not able to see a COM port on the Tools menu when I press BOOTSEL while plugging in the pico. When I do that, pico mounts as a drive and I can see that in Device Manager. It refuses to show up on the Tools menu in this mode. 

To replicate the issue, I drag and drop the Flash_nuke.uf2 to clean out the board and then replug it by pressing down on BOOTSEL button. It shows up perfectly as a drive but not on the Tools menu on the IDE. It doesn't show the PORT option
But if do the same on a second PC of mine(Win 11, same IDE and core) and follow the same steps, it shows up on the Tools-->Port menu under uf2conv ports as a UF2 Board. This enables me to upload Blink sketch directly to pico from IDE. Once a code has been uploaded, I can plug this pico board back into the first PC and it shows up as a COM port on the IDE of the First PC. I still cannot upload via IDE by selecting this COM Port.  It fails to upload with the following error

`Failed uploading: uploading error: exit status 1 Resetting COM12
Converting to uf2, output size: 134656, start address: 0x2000
Unable to build drive list`

I'm pretty sure it's something to do with the First PC and not something wrong with the Arduino-Pico core. But I am failing to understand how to resolve this issue on the first PC. Works perfectly well on PC2 and not on PC1. I use PC1 for most Arduino style developments.

Can you please point out where the issue might be to fix this if you have experienced the same?

earlephilhower, 2023-01-16:
Did you install the IDE using the Windows Store?  As documented, that won't work because the permissions they give the IDE are too low to allow it to access the port properly (FWIW the Ubuntu Snap and RedHat? FlatPak prepackaged installs have the same problem in Linux).  IF so, uninstall the IDE from the Win Store and d/l and install the IDE from https://arduino.cc

dev-000, 2023-01-17:
IDE was the one I downloaded from the Arduino website and not from the WIndows store. That's the same one I use for all Arduino development for SAMD21, SAMD51, ESP32 etc. 

I was trying a few more things to check what is exactly happening. In PC1 (Win 10), I have Arduino IDE 1.18 installed normally and yesterday for testing, I was using Arduino 2.03 IDE as a zip folder(Not installed via installer). The issue I mentioned in the prev post was using the Arduino 2.03 IDE.

Here is what I did. I flashed the flash_nuke.uf2 on the pico board to establish a baseline.

I pressed BOOTSEL and plugged in PC1. Pico opened in the mass storage mode and shows up as a drive. Then I opened Arduino IDE 1.18 and saw that the Port option in Tools menu is Greyed out. I closed the IDE and opened up Arduino IDE 2.03, Here I can't see the port in the Tools section at all. So both IDEs work the same and both cannot see the port in mass storage mode. 

Now I plug the pico into PC2(Win 11) and open up IDE 2.03 on PC2. It shows up on the Tools-->Port menu under uf2conv ports as a UF2 Board and I can upload the blinky code to it. Once upload is a success, a code with Arduino-pico core with auto-reset is present on the pico board.

If I plug back this board with code on the PC1(without pressing BOOTSEL), on both IDE 1.18 and IDE 2.03, the COM port shows up on the Tools menu. Now here is where the difference occurs, I can flash a code from PC1 IDE 1.18 and it uploads fine. But when I use IDE 2.03 and press upload, I get the same uploading error I mentioned in prev post. I would need to replug the device to see the COM port again. 

So on PC1, there are 2 problems,

- I cannot see the board in the COM port in IDE 1.18 and 2.03 in mass storage mode.
- I cannot upload code even with an Arduino-pico core with IDE 2.03

On a side note, will I be able to get a .uf2 file if I compile via an Arduino IDE with the arduino-pico core? If so, where can I find that .uf2 file post compilation on the IDE?

dev-000, 2023-01-26:
I have fixed this issue today. Seems the problem in PC1 was that the powershell command was missing from System Environment Path Variable. Seems `uf2conv.py` in [line 238](https://github.com/earlephilhower/arduino-pico/blob/master/tools/uf2conv.py#L238) is calling powershell. That was throwing the `Unable to build drive list` error because it couldn't run the powershell command.

I found a reference for this fix [here](https://samuelpinches.com.au/hacking/problem-uploading-sketch-to-raspberry-pico-rp2040-in-arduino-ide-unable-to-build-drive-list/). But I didn't have to add all the items mentioned in the link to my Environment Path Variable. All I needed to do was add the following line as a new entry to Path Variable on Windows 10, which seems to be the root cause.

`C:\Windows\System32\WindowsPowerShell\v1.0\`

Had to Restart the PC and then upload works fine on both IDEs 1.18 and 2.03 via the `arduino-pico` core.

closed 2023-01-26 by dev-000

earlephilhower, 2023-01-26:
Thanks for the update!  I think this might warrant a better error message.  The current "can't make drive list" one is factually correct but not so helpful, I think you'd agree.

dev-000, 2023-01-26:
Yup. Worth letting the user know via the error message. I had spent some amount of time trying out a bunch of things to figure out what was the problem.

earlephilhower, 2023-01-26:
#1141 adds in a meaningful error in this case.  Again, thanks for following up with your solution!

pinchies, 2023-01-28:
Thanks @earlephilhower and @dev-000 -- that should help users in cases like [this](https://github.com/earlephilhower/arduino-pico/discussions/558) and [this.](https://github.com/earlephilhower/arduino-pico/discussions/337)