earlephilhower/arduino-pico #1127: no member named 'availableSpaceForWrite';

stooged, 2023-01-18:
getting an error compiling a sketch that uses SD.h on older versions and the latest version 2.7.1

```
\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\SDFS\src/SDFS.h: In member function 'virtual int sdfs::SDFSFileImpl::availableForWrite()':
\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\SDFS\src/SDFS.h:279:31: error: 'using element_type = class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'; did you mean 'availableForWrite'?
  279 |         return _opened ? _fd->availableSpaceForWrite() : 0;
      |                               ^~~~~~~~~~~~~~~~~~~~~~
      |                               availableForWrite
```

changing line 279 in SDFS.h to...
```
        return _opened ? _fd->availableForWrite() : 0;
```
allows compile and allows me to use the sd card but you may want to check if it still functions as expected

earlephilhower, 2023-01-18:
Can you give a small sketch that reproduces the failure?  (FWIW, I think your change will cause an infinite loop as `availableForWrite` will call itself until the stack is exhausted...)

stooged, 2023-01-18:
the sd card example produces the error for me
https://github.com/earlephilhower/arduino-pico/tree/master/libraries/SD/examples/Files

this is the error i get compiling that sd files example

```
In file included from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\SD\src/SD.h:24,
                 from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\SD\examples\Files\Files.ino:21:
C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\SDFS\src/SDFS.h: In member function 'virtual int sdfs::SDFSFileImpl::availableForWrite()':
C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\SDFS\src/SDFS.h:279:31: error: 'using element_type = class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'; did you mean 'availableForWrite'?
  279 |         return _opened ? _fd->availableSpaceForWrite() : 0;
      |                               ^~~~~~~~~~~~~~~~~~~~~~
      |                               availableForWrite
Multiple libraries were found for "SD.h"
 Used: C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
 Not used: C:\Users\User\Documents\Arduino\libraries\SD
Multiple libraries were found for "SdFat.h"
 Used: C:\Users\User\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork
 Not used: C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\ESP8266SdFat
exit status 1
Error compiling for board Raspberry Pi Pico W.

```

could the issue be related to the sdfat library?
i just noticed it is using a **SdFat_-_Adafruit_Fork** 

```
 Used: C:\Users\User\Documents\Arduino\libraries\SdFat_-_Adafruit_Fork
 Not used: C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.7.1\libraries\ESP8266SdFat
```

closed 2023-01-18 by stooged

earlephilhower, 2023-01-18:
Nice debug.  Thanks for digging into it!

pschatzmann, 2023-03-04:
Same issue when the (original) greiman/SdFat library has been installed.

troisiemetype, 2023-03-09:
Same issue too with the original SdFat library. I've removed it from the Arduino library folder, it solved the problem.

It's more a problem linked with how the IDE choose one library against another. The counterpart of not having to maintain a makefile. :)