earlephilhower/arduino-pico #1129: Custom USB VID/PID

dev-000, 2023-01-20:
Hi,

I have been trying to play around with setting custom VID/PID on Pico board from Arduino IDE. Seems the arduino-pico core doesn't seem to support any other VID than the Official Raspberry Pi's VID `0x2E8A`. I know VIDs are to be bought and registered from USB Foundation. But folks who have a custom VID should be able to use them with the core. I have read a similar [issue](https://github.com/earlephilhower/arduino-pico/issues/581) in the repo. I thought the fix was simple by editing `RP2040USB.cpp`.  

Following are the changes I did, 

- In order to add a new custom compilation flag for changing VID, I added a line in `makeboards.py` 
`print("%s.build.usbvid=-DSERIALUSB_VID=%s" % (name, vid))` just before the line  `print("%s.build.usbpwr=-DUSBD_MAX_POWER_MA=%s" % (name, pwr))`
- I created a custom board on `makeboards.py` with random VID/PID pair. Ran it and generated a new boards.txt file with `build.usbvid` property 
- Now in the `platform.txt` file, I added a new element `{build.usbvid}` in the compile` C, C++, S` files right after `{build.usbpid}` so that this compilation flag is executed.
- In `RP2040USB.cpp`, I added the following replacing, `#define USBD_VID (0xE60C) // Raspberry Pi` so that functionality is same as VID
```
#ifdef SERIALUSB_VID
#define USBD_VID (SERIALUSB_VID)
#else
#define USBD_VID (0x2E8A) // Raspberry Pi 
#endif
```
Now I compiled a blinky program in Arduino 2.03 IDE and flashed the board, the custom VID/PID is set on the board and I confirmed that in the Windows device manager.

Now here is the issue. After this, to play around, I changed the boards.txt manually by changing `build.usbvid=-DSERIALUSB_VID=` and `build.usbpid=-DSERIALUSB_PID=` to some other random VID/PID pair to see if it's working. After saving the boards.txt and flashing the blinky code again, Pico seems to still show the old VID/PID pair and it's not taking the new pair, So I am guessing, the VID logic which I implemented is not fully correct. In my understanding, this should ideally work, but it is not working always. 

Is there anything that I missed in this implementation? I am not very sure of the compilation workflow of the Arduino core. Ideally, if I change the values( `DSERIALUSB_PID` and `DSERIALUSB_VID`) it should get picked up by the compiler and VID should have been set. 

Can you kindly suggest if this approach is wrong?

earlephilhower, 2023-01-23:
Doing as you suggest should work in theory.  Since the USB licenses those VIDs by vendor, I've hardcoded it to the RPI Foundation to avoid issues, but your change should be fine to override it manually.

W/o all your changes it's hard to say what is wrong, but really if you are doing this on your own with a legit code why not just hardcode the VID/PID in the source file you edited by placing them there directly?  
https://github.com/earlephilhower/arduino-pico/blob/ecaa2bd778eb46fb4e9569f55bc07e693b6256cb/cores/rp2040/RP2040USB.cpp#L96-L97

You could also just use Adafruit_TinyUSB and implement your own USB setup/run code directly using that.  I haven't used it myself, but I would imagine it's simple to set VId/PID programmatically.

closed 2023-01-23 by earlephilhower

dev-000, 2023-01-25:
@earlephilhower Thanks for the response. 
I can change the IDs to hard code USBD_VID and USBD_PID on `RP2040USB.cpp` file as you mentioned but these would be hardcoded for every board then.
 
I resolved this issue today, I was changing PID/VID in the boards.txt file manually and I was flashing it with the IDE without restarting it 🤦‍♂️ So it was taking the old PID/VID. I tested it now editing the boards.txt, restarting the IDE and then flashing it. It works fine.

Should I add this to this repo? The thing is current repo doesn't support changing VIDs. Take Adafruit's Feather board, for example, they have a different VID if we compile it with this core as is, VID will be replaced with RPi's VID even though Adafruit's VID is mentioned in the `makeboards.py` file.

If this needs to be put up as an addition to the current repo, I can fork it and raise a PR. I think the files that would need change will be `makeboards.py` for adding a new flag, `boards.txt` will get updated for all boards with this flag, `platforms.txt` for adding the compilation flag and `RP2040USB.cpp for the preprocessor directives.

earlephilhower, 2023-01-27:
So if I understand it, we've got a bug where the VID is ignored completely by the core?  If so, a PR would be much appreciated fixing it!

Also, maybe a simpler way to override the VID/PID may be to define `USB_OVERRIDE_PID` and `USB_OVERRIDE_VID` and change `RP2040USB.cpp` to use those when defined.  A user can then, in their `platform.local.txt` do:
````
compiler.cpp.extra_flags=-DUSB_OVERRIDE_VID=0xabcd -DUSB_OVERRIDE_PID=0x1234
````

Those could also be passed in thru Platform.IO methods covering all bases...

dev-000, 2023-01-27:
I am not sure if adding a `platform.local.txt` is the right approach because currently PIDs are changed in a particular way so I was just following the same for VIDs with a separate compiler flag too. Since a user is adding a VID on the `makeboards.py` file, I would assume that they would expect it to be compiled via that VID. 

TBH, I think the preprocessor section for `SERIALUSB_*` can entirely be removed from `RP2040USB.cpp` as you would always be adding the compile flags. I am also not sure why separate `SERIALUSB_PID` was created when we could have potentially used `-DUSB_VID` and `-DUSB_PID` for doing this and reduce the compile flags on the `platforms.txt`

Since I don't know what will break if we do it via the methods mentioned above, I went with my old approach. I forked the project and did a commit in my repo for those changes I mentioned. You can check it [here](https://github.com/dev-000/arduino-pico/commit/17079d8d80295de9c829656a1b2d9a92aabc73d5) when free. I didn't put up a PR yet. I am not that familiar with Platform.IO so I would not be able to test it out on my end. I am assuming we would need to do a change for adding the VID flag [here](https://github.com/earlephilhower/arduino-pico/blob/master/tools/platformio-build.py#L218) but I am not sure.

earlephilhower, 2023-01-27:
Ah, I think see.  You were just not able to use the standard (non-RPi) VID that's defined in the core for things like the Adafruit boards/etc., yes?  If so then the override isn't needed, just the initial fix.  I'll give it a go this weekend and update...

dev-000, 2023-01-28:
> Ah, I think see. You were just not able to use the standard (non-RPi) VID that's defined in the core for things like the Adafruit boards/etc., yes? 

Exactly. VID was not taken from what's mentioned on the `boards.txt`.
From a custom board manufacturer's POV, if they want to add a new board to the `arduino-pico` core, they should ideally only add a line to the `makeboards.py` file with the board-specific details and add a new folder (in the master `variants` folder) with the same `name` field with their pin specifications edited in `pins_arduino.h`. Then they raise a PR and that's it. This might help smoothen out the future PR's for adding custom boards if we can add the steps to the Readme section also. As I am sure more and more custom boards will be created with this core as it's much more popular than the "Official Mbed OS" one.
 
While on the topic of USBs, one other low-hanging [issue](https://github.com/earlephilhower/arduino-pico/blob/ecaa2bd778eb46fb4e9569f55bc07e693b6256cb/cores/rp2040/RP2040USB.cpp#L313) that could be fixed is that Manufacturer and Product Names were hard coded. It was also not taken from the `boards.txt` even though the board manufacturer name and product name are entered on the `makeboards.py` file. If a custom board is plugged in, the product name is the one which pops up as a notification on Windows as "Setting up XYZ". Currently, XYZ defaults to "PicoArduino".

On a side note on PIDs, (I think need to put this as a feature request, documenting here as its related) the core doesn't show the name of the custom board in brackets on `Tools--> Port --> COM` section as other cores like SAMD or the official Mbed Pico core. Check the images below. This info is taken from VID/PID pair from `boards.txt`. Check the images below for an RPi Pico board connected on the COM port with the `"Official" Mbed OS` core and `arduino-pico` core.

[image]

[image]

Extending that thought, `Tools--> Get Board Info` also returns VID/PID as N/A on the `arduino-pico`. But this one might be much bigger fix.

earlephilhower, 2023-01-28:
@dev-000 , thanks for reporting this.  I'm surprised nobody has mentioned it before as it seems pretty basic!

#1144 should take care of all your comments, including the port name and get more info (after flashing the board once).  I've just tested an RPI Pico board flashed as an Adafruit  Feather and see:
[image]
and
[image]

dev-000, 2023-01-29:
@earlephilhower Thanks for the fix. I pulled the latest and it seems to be working but with some issues for me. 
I am unable to upload a code twice on a Rpi board. IDE uploads fine once and doesn't seem to reinitialise the port after the upload. So I can't see any port on the Tools menu after an upload. Although the device manager seems to show the board as connected. 
Also, port name in brackets and get more info is also not working for me. I am slightly short on time to do a full test today as to why this might be the case in Windows. Not sure if others are facing the same issue on Windows.

earlephilhower, 2023-01-29:
That's a normal Windows thing.  Now that the USB VID/PIC are different you need to use the usual Zadig/etc. since this is a"new" device type.

dev-000, 2023-01-29:
I tested it in detail right now. Seems VID PID changes are a success with the latest commit. I can see the VID and PID changes are reflected in the Device Manager on Windows 10 when different boards are compiled.

I am still not seeing the port name in brackets. Get Board Info is also not working for me in Win10 IDE 2.0.3. You may want to check with someone else on Win 10 if it's fixed for them. 
[image]

[image]

Also, hardcoding the USB product [name](https://github.com/earlephilhower/arduino-pico/blob/ecaa2bd778eb46fb4e9569f55bc07e693b6256cb/cores/rp2040/RP2040USB.cpp#L313) is causing the following popup on Windows. This does make it confusing for a user as the user wouldn't know the difference between an official Pico and a Feather Pico. These parameters are passed anyway on the `makeboards.py`.
[image]

Regarding the upload issue I mentioned above, it seems to be a by-product of the other [issue](https://github.com/earlephilhower/arduino-pico/issues/1123) I was facing.  The new error [message](https://github.com/earlephilhower/arduino-pico/pull/1141) is getting thrown sometimes but I am not able to pinpoint when or why it occurs and why it's random. Restarting the PC and clearing the temp IDE cache, seems to fix it. I will need to repeat a few more tests to ascertain the root cause. I will update it here if I am able to figure it out.

earlephilhower, 2023-01-29:
Thanks for the detailed info, but I think this is all IDE related and out of scope here.

The Board Name and ID are all based off of the USB VID/PID via a table in the `boards.txt` file.  If the Pico is exposing the right VID/PID then the problem is related to IDE version 2.0.  I just ran 1.18 and all seems fine:
[image]
[image]

If it's really bugging you, I'd suggest opening something up on the Arduino repo here.  They have been doing good fixes in the 2.0, but there's still a little way to go (i.e. no plug in support yet, for example)

dev-000, 2023-01-29:
Yes. You are right. I just installed 1.8.19 and it seems to work fine for the `arduino-pico` core. Could be IDE 2.x related. 
But all other boards on SAMD series seem to show up fine on IDE 2.0.3 along with the official Mbed OS Pico core. I am trying to see what the difference is for the `arduino-pico` core on the IDE 2.0. 
This one is not much of a pain point as the USB product name bug not showing up on Windows. This one just helps when multiple boards are plugged in and you need to know which is which without unplugging and checking.