earlephilhower/arduino-pico #1134: Regarding low power mode/power saving operation

ronie19, 2023-01-22:
I am using RPi Pico with SIM868 GPS/GSM module with bare minimum connections and is powered using a Lithium Ion battery. 
My system sends an SMS, waits for 15 minutes and sends SMS again and waits for 15 mins and so on in a loop.
I am using millis method to create delay. What is the best way to conserve power for the Pi Pico in battery driven system where program is in continuos loop.

 ( I am not worried about SIM868 power consumption.)

Phoenix1747, 2023-01-22:
You could use the [Adafruit SleepyDog Arduino Library](https://github.com/adafruit/Adafruit_SleepyDog), it officially supports the rp2040 architecture. Although, in my tests it didn't really go into sleep (steady power consumption), so it looked more like it just waits much like an interrupt. Maybe I made some mistakes, so YMMV.

earlephilhower, 2023-01-22:
I think the Adafruit library doesn't actually enable any low power mode.

Right now in the core the is no low power support but you are able to use the native SDK calls.  (My DVM in ammeter mode isn't very accurate at microwatts, so I've not looked into it too deeply.)

Check the pico-extras repo from Raspberry Pi where they have a RTC sleep demo.  You can power down or stop clocks to most units in the SOC, but on wake up it's up to you too make sure they're reenabled and configured properly.  I think you'll need to do some work on the core as well because right now it assumes when it sets something up, like the ADC or PWM, that nothing is going to go and reconfigure it...