earlephilhower/arduino-pico #1138: Custom PIO support

aidanchandra, 2023-01-25:
Hello!

I hope all is well. I was perusing the documentation and only found a few instances mentioning PIO blocks, most of which were pre-implemented. The only mention of custom PIO blocks directs to an assembler, but doesn't provide guidance on how to run custom PIO blocks.

I'm using PlatformIO with this, and it's run spectacularly thus far. I just need to know if there is support for custom PIO blocks. And if so, how to compile and inject/run them.

Thank you in advance

Aidan

earlephilhower, 2023-01-25:
You follow the same workflow for PIO use as described in the SDK docs.  PIOASM is shipped, you'll build a .h from the .pio, and then write your init/deploy/etc. functions just like you'd do w/the raw SDK.  The core itself doesn't hardcode any assumed PIOs, SMs, or program memory locations, so if you use the SDK it should coexist just fine w/the core.  [There **may** be some hardcoded dependencies in the SDK for the Pico W's PIO-based SPI, so be careful there and init the WiFi before adding your PIO programs]

There is some work on including an auto `.pio`->`.pio.h` build step in the PIO build script, please check https://github.com/platformio/platform-raspberrypi/pull/36#issuecomment-1326145816

Good luck!

closed 2023-01-25 by earlephilhower

BorisBrock, 2023-02-21:
I'm not sure I fully understand your answer. I have created a basic Pi Pico Project with PlatformIO using the Arduino framework.
Which steps are required to add and use a .pio file to my project?

earlephilhower, 2023-02-21:
If the PIO build script has not yet included `pioasm`, then you just manually compile the `.pio` to a `..pio.h` using `pioasm` from the command line once, and then include that file in your code.  It's straightforward, and how the PIO is used in the core (manual compiles).