earlephilhower/arduino-pico #1143: Sketch -> Export compiled binary: export both .uf2 and .bin

cryogentail1, 2023-01-28:
Hi all,
the menu command "Export compiled Binary" copies from the build folder only the .UF2 file.
This is usefull for programming directly but I need also the .bin file to use it for OTA updates (via http download).
Now I have to copy by hand :-(
Thanks for any suggestion.
  Nicola

JAndrassy, 2023-01-28:
just put this file next to platform.txt in your installation
https://github.com/JAndrassy/ArduinoOTA/blob/master/extras/rp2040/platform.local.txt

earlephilhower, 2023-01-28:
AFAIK, There is no support in the IDE for getting more than one file from the menus.

I *think* you can get a BIN instead of a UF2 by adding a line to your `platform.local.txt`
````
build.preferred_out_format=bin
````

Alternatively, you can use the info in https://learn.sparkfun.com/tutorials/wireless-arduino-programming-with-electric-imp/getting-hex#:~:text=By%20default%2C%20Arduino%20outputs%20the,hex%20 to grab the BIN from an easy to find, well known spot.

closed 2023-01-28 by earlephilhower

cryogentail1, 2023-01-28:
> AFAIK, There is no support in the IDE for getting more than one file from the menus.

This IDE has so many pros and cons...

> 
> I _think_ you can get a BIN instead of a UF2 by adding a line to your `platform.local.txt`
> 
> ```
> build.preferred_out_format=bin
> ```

Yes, this works.

> Alternatively, you can use the info in https://learn.sparkfun.com/tutorials/wireless-arduino-programming-with-electric-imp/getting-hex#:~:text=By%20default%2C%20Arduino%20outputs%20the,hex%20 to grab the BIN from an easy to find, well known spot.

I know where are the compiled files but it's contrary to my philosopy to do by hand a (very simple and repetitive) thing that the computer can do automatically!
Thanks :-)
       Nicola

earlephilhower, 2023-01-28:
Yup, automate all the things! is a great motto to live by. :)

`arduino-cli` will let you compile from the CLI and automate stuff like that, if you really are having issues.  It's what the newer IDE calls to do the build process, anyway.  Platform.IO may also help here (but I'm not an expert and someone else will need to chime in on how to get well known BINs)...