earlephilhower/arduino-pico #1149: PlatformIO: upload to Adafruit Trinkey RP2040 QT errors out

salfter, 2023-01-31:
Maybe I'm doing something wrong, but I've tried uploading both with and without holding in the BOOTSEL button and I keep getting an error.  With BOOTSEL pressed, it usually fails like this:

```
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
```

Without pressing BOOTSEL, I get this:

```
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Auto-detected: COM5
Forcing reset using 1200bps open/close on port COM5
Uploading .pio\build\pico\firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
```

I can copy firmware.uf2 from .pio/build/pico/ onto the device and get my code running that way, so this isn't a showstopper, but it'd be nice to have the upload button working as it does for the other platforms I use.

## Steps to duplicate

Create a new PlatformIO project for the Raspberry Pi Pico, and override the settings in platformio.ini as follows:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = adafruit_trinkeyrp2040qt
framework = arduino
board_build.core = earlephilhower
```

The code I'm testing with is a mouse wiggler that I can change over to a mouse clicker, though I suppose a "blinky" program or anything known to work could be used for main.cpp:

```
#include <Arduino.h>
#include <Mouse.h>

void setup() {
  Mouse.begin();
}

void loop() {
  Mouse.move(-5,0,0);
  delay(1000);
  Mouse.move(5,0,0);
  delay(1000);
  // Mouse.click();
  // delay(30000);
}
```

maxgerhardt, 2023-02-02:
>I can copy firmware.uf2 from .pio/build/pico/ onto the device and get my code running that way, 

That may have to be done initially, but starting from that, it should work through the reset-via-1200bps-touch. 

DId you do the procedure where you plug the board in in BOOTSEL mode and use Zadig to load WinUSB or libusb-win32 drivers on the "RP2 Boot2 (Interface 1)" deveice like [this sasy](https://github.com/platformio/platform-raspberrypi/issues/2#issuecomment-828586398)? Otherwise rp2040load won't be able to access the USB device..

salfter, 2023-02-02:
> DId you do the procedure where you plug the board in in BOOTSEL mode and use Zadig to load WinUSB or libusb-win32 drivers on the "RP2 Boot2 (Interface 1)" deveice like https://github.com/platformio/platform-raspberrypi/issues/2#issuecomment-828586398? Otherwise rp2040load won't be able to access the USB device..

I didn't know about that, but then I usually do this sort of stuff under Linux, not Windows.  Once I did that, programming behaved as expected.  Thanks!

closed 2023-02-02 by salfter