earlephilhower/arduino-pico #1152: SPI 16-bit Receving is Not Working

vishnumaiea, 2023-02-01:
I am trying to communicate with an IC called **CJ125** through RP2040's SPI. CJ125 sends and receives **16-bit data**. So I configured the `SPI0` port (the default one) and tried to send a sample command `0x7800` using the `transfer16()` call. The code that implements that function is below.

```cpp
#define   PIN_SPI_CIPO      4
#define   PIN_SPI_COPI      7
#define   PIN_SPI_SCK       6
#define   PIN_SPI_CS        5

uint16_t _writeCJ125 (uint16_t TX_data) {
  Serial.println ("Writing to CJ125..");

  digitalWrite (PIN_CJ125_CS, LOW);

  //Set chip select pin low, chip in use.
  SPI.beginTransaction (SPISettings (125000, MSBFIRST, SPI_MODE1));

  //Transmit request.
  uint16_t response =  SPI.transfer16 (TX_data); // TX_data == 0x7800

  SPI.endTransaction();

  //Set chip select pin high, chip not in use.
  digitalWrite (PIN_CJ125_CS, HIGH);

  Serial.print ("Returning response: 0x");
  Serial.println (response, HEX);
  return response;
}
```

I am able to send the data (`0x7800`), and CJ125 responds with the expected response of `0x2855`. BUT, I am not able to get that data as a return of the `transfer16()` function. It always returns 0.

The reason why I know that the CJ125 is sending the correct response is that I connected my Logic Analyzer to the SPI pins and I can capture the command and response from the interface. Below is what I captured from the SPI bus.

[image]

So for some reason, Arduino-Pico's SPI implementation is not able to return the 16-bit data it is receiving on the **CIPO** pin. I am using `GPIO4` as the **CIPO** and it is a valid assignment.

Since the RP2040 was not able to read the incoming data or it is being lost for some reason, I tried implementing the same with the **Pico C/C++ SDK**. The code is below.

```cpp
int main() {
  // Enable USB serial so we can print
  stdio_init_all();
  sleep_ms (2 * 1000);
  printf ("CJ125 SPI Communication\n");

  gpio_init (PIN_SPI_CS);
  gpio_set_dir (PIN_SPI_CS, GPIO_OUT);

  spi_init (spi_default, 1 * 125000);
  spi_set_format (spi_default, 16, SPI_CPOL_0, SPI_CPHA_1, SPI_MSB_FIRST);

  // Assign SPI functions to the SPI pins
  gpio_set_function (PIN_SPI_CIPO, GPIO_FUNC_SPI);
  gpio_set_function (PIN_SPI_SCK, GPIO_FUNC_SPI);
  gpio_set_function (PIN_SPI_COPI, GPIO_FUNC_SPI);
  // gpio_set_function (PIN_SPI_CS, GPIO_FUNC_SPI);

  uint16_t out_buf16 [10];
  uint16_t in_buf16 [10];

  out_buf16 [0] = 0x7800;
  in_buf16 [0] = 0;

  while (1) {
    printf ("Sending command 0x%x to CJ125..\n", out_buf16 [0]);

    gpio_put (PIN_SPI_CS, 0);
    spi_write16_read16_blocking (spi_default, out_buf16, in_buf16, 1);
    gpio_put (PIN_SPI_CS, 1);

    if (in_buf16 [0] != 0) {
      printf ("Response received: 0x%x\n", in_buf16 [0]);
      in_buf16 [0] = 0;
    }
    else {
      printf ("No response received\n");
    }

    sleep_ms (1 * 1000);
  }
}
```

Surprisingly, this code works just fine as seen from the serial monitor screenshot below.

[image]

Thinking it must some issue with the Arduino-Pico's SPI implementation, I tried adding the same C/C++ SDK code to my Arduino sketch (using the native SPI initialization), and guess what, it was not working. So my assumption is that something that is specific to Arduino-Pico is interfering with the SPI communication. I am hoping someone can shine some light on this issue 😅

I am using the latest version of the Pico C/C++ SDK v1.4.0 and Arduino-Pico version 2.7.1.

earlephilhower, 2023-02-01:
Did you set the SPI pinout using the `SPI.setRX(PIN_SPI_CIPO);`  somewhere before `SPI.begin()`?

The default pinouts may not match the ones you're requesting, so it is probably shifting in another pin to the device and returning those 0s...

vishnumaiea, 2023-02-01:
I also tried enabling the SPI debugging messages and got the below messages on the serial monitor.

[image]

earlephilhower, 2023-02-01:
You're using HW chip select with the incomplete snipped you pasted above, and I don't see the CS pin being set to `pinMode(xxx, OUTPUT)` before you write to it.

You need to `SPI.begin(false)` as documented to use SW chip select.  And all pins come up as input-only IIRC so the CS digitalWrites you're doing probably aren't making it off chip.

Bodmer, 2023-02-02:
In the original post the CS pin is listed as:
#define   PIN_SPI_CS        5

Yet the original post ed code uses a different CS pin:
digitalWrite (PIN_CJ125_CS, LOW);

The working SDK code seems to use the correct CS pin:
gpio_put (PIN_SPI_CS, 0);

So it is likely that this is where the problem is.

You can run the following code for example and with pin 4 linked to 7 the correct value is read from the bus:
```
#include <SPI.h>

#define   PIN_SPI_CIPO      4
#define   PIN_SPI_COPI      7
#define   PIN_SPI_SCK       6
#define   PIN_SPI_CS        5

void setup() {
  SPI.setRX (PIN_SPI_CIPO);
  SPI.setSCK (PIN_SPI_SCK);
  SPI.setTX (PIN_SPI_COPI);
  SPI.begin();
}

void loop() {
  Serial.println ("Writing...");

  digitalWrite (PIN_SPI_CS, LOW);

  //Set chip select pin low, chip in use.
  SPI.beginTransaction (SPISettings (125000, MSBFIRST, SPI_MODE1));

  //Transmit request.
  uint16_t response =  SPI.transfer16 (0x7800); // TX_data == 0x7800

  SPI.endTransaction();

  //Set chip select pin high, chip not in use.
  digitalWrite (PIN_SPI_CS, HIGH);

  Serial.print ("Returning response: 0x");
  Serial.println (response, HEX);
}
```
This suggests that using the wrong CS line is indeed the problem.

vishnumaiea, 2023-02-02:
Thanks for your inputs. It was a bad idea to prune the code before posting, because I just tested the code below and can confirm it is **working**.

```cpp

//==============================================================================//

#include <SPI.h>

#define   PIN_SPI_CIPO      4
#define   PIN_SPI_COPI      7
#define   PIN_SPI_SCK       6
#define   PIN_SPI_CS        5

//==============================================================================//

void initSPI() {
  Serial.println ("Initializing SPI..");

  SPI.setRX (PIN_SPI_CIPO);
  SPI.setSCK (PIN_SPI_SCK);
  SPI.setTX (PIN_SPI_COPI);
  // SPI.setCS (PIN_SPI_CS);  // Not used
  SPI.begin();
}

//==============================================================================//

void setup() {
  Serial.begin (115200);
  delay (2000);

  Serial.println ("CJ125 SPI Test");
  delay (1000);

  pinMode (PIN_SPI_CS, OUTPUT);
  digitalWrite (PIN_SPI_CS, HIGH);

  initSPI();
}

//==============================================================================//

void loop() {
  Serial.println ("Writing to CJ125..");

  digitalWrite (PIN_SPI_CS, LOW);

  SPI.beginTransaction (SPISettings (125000, MSBFIRST, SPI_MODE1));

  //Transmit request.
  uint16_t response =  SPI.transfer16 (0x7800);

  SPI.endTransaction();

  digitalWrite (PIN_SPI_CS, HIGH);

  Serial.print ("Response received: 0x");
  Serial.println (response, HEX);

  delay (1000);
}

//==============================================================================//
```

So the issue was that I was making a call to initialize the I2C interface with default pins in my "original code" 😅.

```cpp
Wire.begin (I2C_PERIPHERAL_ADDR);
```

This line came after the SPI initialization. That means, the initialization caused a **pin assignment conflict** by reassigning the GPIOs **4** and **5** (the same pins I was using for the SPI) for the I2C interface 🤦🏻. That was causing the issue. The macro `PIN_CJ125_CS` is assigned to `PIN_SPI_CS` in my original code BTW.

It was my fault for not isolating just the SPI communication and test it, which I only did now after seeing the code from @Bodmer 🫠

I also captured waveforms of the CIPO line in both the working case and non-working case, using a DSO and observed a tiny difference between them.

[image]
Image 1: CIPO line in working case

[image]
Image 2: CIPO line in non-working case

There was a small positive offset of around **+0.8V** in the non-working case. I had noticed this and had some **gut feeling** about it. But it never turned the light bulb on. I will be careful to isolate problems and test them separately in the future.

closed 2023-02-02 by vishnumaiea

earlephilhower, 2023-02-02:
Awesome, thanks for the update!