earlephilhower/arduino-pico #1153: Bogus character received on Serial1 after .begin()

dlkeng, 2023-02-02:
**Raspberry Pi Pico / RP2040 Version 2.7.1**

On a Raspberry Pi Pico, on **Serial1**, shortly after ```Serial1.begin()```, a single bogus character is received by the UART1. This character is dependent on the baud rate: At 115200, the character is `0xFC`; at 38400 and 57600, the character is `0xFF`. At less than 38400, no bogus character is received.

This has been seen on two different Raspberry Pi Pico's and one Raspberry Pi Pico W.
It was seen on GP0/GP1, GP12/GP13, and GP16/GP17 pins associated with UART1.
This does **not** occur if the Rx pin is tied low or floating. It only occurs when the Rx pin is tied high (pullup to +3V3 or direct to +3V3) or driven high by an RS-232 transceiver.

This does **not** occur on **Serial2**.

This does **not** occur on the official ```Arduino Mbed OS RP2040``` core.

The following is a simple test sketch to demonstrate this issue:
```
void setup()
{
    Serial1.begin(115200);
}

void loop()
{
    if (Serial1.available())
    {
        // At less than 38400, no characters at startup!
        // 38400, 57600: emits one 0xFF at startup!
        // 115200: emits one 0xFC at startup!
        Serial1.print(Serial1.read(), HEX);
    }
}
```

earlephilhower, 2023-02-02:
Sorry, can't reproduce this at all here with a floating input (probably the worse case).  Could you please triple check you're running the latest core?

In earlier versions an invalidly framed byte would be returned as `available()`, but that was fixed and tested pretty extensively by myself and others in #1021 / #1036 .  If it's reporting data, then it appears that somehow a vaildly framed byte is showing up on the pins.  Can you grab a logic analyzer trace?

dlkeng, 2023-02-02:
I just got Version 2.7.1 today. Also, the issue does **not** occur with a floating Rx input, only an Rx input that is driven or tied high.

dlkeng, 2023-02-02:
I updated to the #1154 version of SerialUART.cpp (and SerialUART.h) and that appears to have corrected this issue.
Thanks.

PR #1154 (2023-02-02): Avoid initial glitch on Serial1/2 when RX high
Fixes #1153

Set up the GPIO redirects before the UART is pulled from reset to avoid a possible bad byte at `Serial1/2.begin()`.
commit b400897c (2023-02-02): Avoid initial glitch on Serial1/2 when RX high (#1154)

Fixes #1153

Set up the GPIO redirects before the UART is pulled from reset to avoid
a possible bad byte at `Serial1/2.begin()`.
closed 2023-02-02 by earlephilhower