earlephilhower/arduino-pico #1157: Is USBHost mode supported?

xployt404, 2023-02-03:
So apparently the pico has host capabilities so it can read input from a keyboard that is attached to the micro USB port via an adapter. The process of doing it with the pico sdk is way to complicated so could you maybe please implement it into this package or is it already? That would really help me out! :)

earlephilhower, 2023-02-03:
Sorry, there are no plans to do a host mode implementation.  Just not enough folks using it, and it requires (as you said) a bunch of add'l code to get running.  Always happy for PRs, though, so if you do develop a library please do share it!

You could also check out TinyUSB.  They *might* have something for this, but I really haven't looked into it.

Good luck!

closed 2023-02-03 by earlephilhower

xployt404, 2023-02-03:
That was actually my plan and then I saw the official full featured easy to read API documentation (sarcasm there is no documentation because as you said not many people use it). I will still try it and issue a pull request when I manage to add such a library.
Thank you anyways for the quick response and have a nice day or evening! :)