earlephilhower/arduino-pico #1180: NO_USB does not allow for fully custom devices

sanjay900, 2023-02-13:
I would like to use NO_USB (with platform.io, so PIO_FRAMEWORK_ARDUINO_NO_USB) so that i can implement a custom vendor device and some host devices with `Pico-Pio_USB`, but it seems that `libpico.a` does not have support for this built in, and i end up with a lot of linker errors and things don't compile, is there a way around this?

```sh
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/pico_ble/src/pico/pico_ble/main.cpp.o: in function `tud_vendor_control_xfer_cb':
main.cpp:(.text.tud_vendor_control_xfer_cb+0x114): undefined reference to `tuh_control_xfer'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: main.cpp:(.text.tud_vendor_control_xfer_cb+0x158): undefined reference to `tuh_control_xfer'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/pico_ble/src/pico/pico_ble/main.cpp.o: in function `loop':
main.cpp:(.text.loop+0x18): undefined reference to `tuh_xinput_ready'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: main.cpp:(.text.loop+0x24): undefined reference to `tuh_xinput_receive_report'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: main.cpp:(.text.loop+0x38): undefined reference to `tuh_task_ext'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/pico_ble/src/pico/pico_ble/main.cpp.o: in function `_Z25send_report_to_controllerPhh':
main.cpp:(.text._Z25send_report_to_controllerPhh+0x10): undefined reference to `tuh_xinput_mounted'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: main.cpp:(.text._Z25send_report_to_controllerPhh+0x20): undefined reference to `tuh_xinput_send_report'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/sanjay/.platformio/packages/framework-arduinopico/lib/libpico.a(hid_device.c.obj): in function `hidd_control_xfer_cb':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c:266: undefined reference to `tud_hid_descriptor_report_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c:300: undefined reference to `tud_hid_get_report_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c:328: undefined reference to `tud_hid_set_report_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/sanjay/.platformio/packages/framework-arduinopico/lib/libpico.a(hid_device.c.obj): in function `hidd_xfer_cb':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c:412: undefined reference to `tud_hid_set_report_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/sanjay/.platformio/packages/framework-arduinopico/lib/libpico.a(msc_device.c.obj): in function `proc_read10_cmd':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:843: undefined reference to `tud_msc_read10_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/sanjay/.platformio/packages/framework-arduinopico/lib/libpico.a(msc_device.c.obj): in function `mscd_xfer_cb':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:476: undefined reference to `tud_msc_scsi_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/sanjay/.platformio/packages/framework-arduinopico/lib/libpico.a(msc_device.c.obj): in function `proc_builtin_scsi':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:658: undefined reference to `tud_msc_test_unit_ready_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:691: undefined reference to `tud_msc_capacity_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:728: undefined reference to `tud_msc_capacity_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:764: undefined reference to `tud_msc_inquiry_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/sanjay/.platformio/packages/framework-arduinopico/lib/libpico.a(msc_device.c.obj): in function `proc_write10_new_data':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:905: undefined reference to `tud_msc_write10_cb'
/home/sanjay/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/sanjay/.platformio/packages/framework-arduinopico/lib/libpico.a(msc_device.c.obj): in function `mscd_xfer_cb':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c:544: undefined reference to `tud_msc_scsi_cb'
```

earlephilhower, 2023-02-13:
You're listing a ton of USB callbacks that your code itself needs to implement if you disable the built-in core USB.  If you don''t need to do a MSD, just bring in the code from here:
https://github.com/earlephilhower/arduino-pico/blob/84206eb237ae2cc966c0eae759ba09c33367bb77/cores/rp2040/RP2040USB.cpp#L383-L457

There are also some other USB callbacks (BLE??) that show up in your code which are not part of any of the core/libpico.a at all.  Nothing we can do here about those, you'll need to implement the CBs as appropriate.

closed 2023-02-13 by earlephilhower

sanjay900, 2023-02-13:
Thank you, that works, however, my other problem is that usb host support is also not enabled, which leads to errors like
```
src/pico/pico/main.cpp:189:17: error: 'tuh_control_xfer' was not declared in this scope; did you mean 'tud_control_xfer'?
  189 |                 tuh_control_xfer(&xfer);
      |                 ^~~~~~~~~~~~~~~~
      |                 tud_control_xfer
src/pico/pico/main.cpp:200:17: error: 'tuh_control_xfer' was not declared in this scope; did you mean 'tud_control_xfer'?
  200 |                 tuh_control_xfer(&xfer);
      |                 ^~~~~~~~~~~~~~~~
      |                 tud_control_xfer
```

earlephilhower, 2023-02-13:
Sorry, I have no idea how the host stuff in TinyUSB works, and it's not supported in the core build of tinyUSB.  Maybe you can enable it using Adafruit_TinyUSB?

sanjay900, 2023-02-13:
I was able to enable it by simply changing `tools/libpico/tusb_config.h` and enabling host support there and then running make-libpico.sh.
I may just do this anyways since i want to use a custom build of tinyusb anyways, thanks for your help!