earlephilhower/arduino-pico #1181: Printf doesn't appear to work

sanjay900, 2023-02-13:


earlephilhower, 2023-02-13:
Thanks, that does look like an issue.  It's a pretty think POSIX shim so should be simple to fix (famous last words).

That said, in the mean time you can simply use `Serial1.printf()` in your code.  `::printf` was added to `Printable` for ease of use and should work on all Serial/File/etc.

earlephilhower, 2023-02-13:
This was very familiar so I looked back and found the problem and a fix which, for some reason, I dropped before including in the toolchain.  Basically the size of a `recursive_mutex` is not the same at the `fp` structure (for `stdout` in this case) so basically we try and lock a garbage mutex.

https://github.com/earlephilhower/pico-quick-toolchain/pull/15

I do not remember at all why I closed it w/o comment, but need to do some testing again to see if there was something else this broke.  End result will probably be a new toolchain spin since the Newlib structs will need to be different.

maxgerhardt, 2023-02-13:
Maybe had something to do with https://github.com/earlephilhower/arduino-pico/pull/677 or https://github.com/earlephilhower/arduino-pico/issues/614#issuecomment-1159690720.

maxgerhardt, 2023-02-13:
This is weird. I tested out different versions of https://github.com/earlephilhower/pico-quick-toolchain/releases and I found that  1.5.0-a crashes with the hardfault, but 1.4.0-c does not crash. 

When I manually download + extract [the archive](https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.4.0-c/x86_64-w64-mingw32.arm-none-eabi-0196c06.220714.zip) and point to it using
```ini
platform_packages =
  toolchain-rp2040-earlephilhower@symlink://C:\Users\Max\Downloads\x86_64-w64-mingw32.arm-none-eabi-0196c06.220714\arm-none-eabi
```
it does not crash.

earlephilhower, 2023-02-13:
Yeah, I'm thinking 1.4.0 had the patch file in the build tree but not in git so the build included the fix.  For 1.5.0 I was careful to clean up anything that wasn't in git, so definitely no patch there.

When the toolchain was started the CI limits wouldn't let it get near complete (compiling GCC 8 times takes a *lot* of CPU cycles!).  Now, I think GH CI has something like a 6hr limit so it might be time to see if I can make the build run in the cloud in a clean VM instead of my server.

#614 was a different thing at the base.  There's a global re-ent struct newlib uses and it needs to be task-specific.  Not really a lock thing, more of a shared struct that has "thread" state.

In any case I'm going to do some more checking w/plain and FreeRTOS w/the revived patch and if it's good will do a commit and toolchain rebuild.

sanjay900, 2023-02-13:
Something very interesting is that if i manually specify the 1.5.0-a toolchain from the releases page, that actually works:
```ini
platform_packages =
   toolchain-rp2040-earlephilhower@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.5.0-a/i686-linux-gnu.arm-none-eabi-5007782.221223.tar.gz
   ```

earlephilhower, 2023-02-14:
Not sure what to say to that, because I just re-ran `get.py` to pull the 1.5.0-a binaries and did the very basic check in GDB showing it's busted:
````
(gdb) print sizeof fp->_lock
$3 = 4
(gdb) print sizeof(recursive_mutex_t)
$4 = 8
````

The FILE->UNIX file number structure has 4 bytes allocated for an 8 byte recursive mutex.  This is because Newlib uses an opaque `LOCK_T` struct that you need to adjust to match your mutex size, and by default it was 4 bytes.  The actual recursive mutex implementation on the Rpi Pico takes 8  bytes.  So, you will either get lucky and the recursive mutex will overwrite another field in the FP struct but work, or unlucky when that following structure member overwrites the last 4 bytes of the recursive-mutex with "garbage" and crash.

earlephilhower, 2023-02-16:
This simple test now works with and without FreeRTOS.  Might need to wait for PIO team to publish the new toolchain, or use the 1.5.0-b ones manually.

````
//#include <FreeRTOS.h>
void setup() {
}
int cnt=0;
void loop() {
  printf("%d\n", cnt++); delay(1000);
}
````

closed 2023-02-16 by earlephilhower

maxgerhardt, 2023-02-16:
Great, I'll test today in PlatformIO and keep an eye on the available versions of the `toolchain-rp2040-earlephilhower` package.

Just for my info, when NO_USB is enabled, does `stdout` go to `Serial1` hardware UART, and when not, it goes to the USB Serial?

earlephilhower, 2023-02-16:
It goes to the debug port specified, or `/dev/null` if none set:

https://github.com/earlephilhower/arduino-pico/blob/7afcec8edce4cfd454321dd6f64f8a2b3779bb83/cores/rp2040/posix.cpp#L36-L46

sanjay900, 2023-02-16:
How long does it usually take for them to appear on PIO?

earlephilhower, 2023-02-18:
I think all the packages just got published.  I lost track at the 40th email from the P.IO team :laughing:

maxgerhardt, 2023-02-19:
Yep, toolchain was updated, and CI is running through.

PR #1196 (2023-02-16): Upgrade to toolchain 1.5.0-b
Includes fixes for ::printf/etc. using stdout/stderr Fixes #1181
commit d1a30094 (2023-02-16): Upgrade to toolchain 1.5.0-b (#1196)

Includes fixes for ::printf/etc. using stdout/stderr
Fixes #1181